foldynl/QLog #11: QLog 0.6.0, Settings: Modes

DL2KI, 2022-03-07:
In the tab "Modes" you find 20 modes. Obviously the classic modes "CW", "LSB", "USB", "RTTY", "DATA" are missing. All available modes are "Disabled" for me, because they are not used.

This leads to the fact that when logging a CW QSO the mode cannot be selected in the QSO window. Whether an automatic takeover of the mode from the rig into the menu fields "Mode" / "Submode cannot be checked.

foldynl, 2022-03-07:
That's strange. Unfortunately, I cannot confirm your finding. v.0.6.0 should have 44 modes where the mentioned modes are present. See screenshot from the release. Was there the same problem observed by you with the previous version (0.5.0)?  

[image]

DL2KI, 2022-03-07:
Here is a screenshot of the window.

[image]

DL2KI, 2022-03-07:
> Have you ever tried to edit a QLog database data file?

No.

> Sorry for spamming.
> 
> I am continuing to investigate the issue but I have a workaround. The table is fully editable therefore you can redefine for example CHIP mode to CW with pre-defined Report format and DXCC should be CW/

No problem. I just tried this once and it seems to work. Tomorrow I will test this further.

According to my memory, this was also the case with version 0.5.0. I just uninstalled version 0.6.0 (purge), and reinstalled version 0.5.0. The settings menu looks the same for both versions. 
 
The database is in the user directory and of course has not changed. I will backup and delete the database tomorrow and check it again.

DL2KI, 2022-03-08:
I have uninstalled version 0.6.0 (purge) and backed up the database "qlog.db". After that I installed version 0.5.0.

After starting QLog the database is regenerated. The new file "qlog.db" has a size of 512,0 KiB.

The Modes menu looks like this:

[image]

When the newly generated database is opened with "SQLite Studio", the content of the "modes, Indices" table looks like this:

[image]

I can't say more about the database, as I have no experience with SQLite or other databases. I installed "SQLite Studio" yesterday only once to have a look at the contents of the database.

After an update to version 0.6.0 nothing changes in the new database.

foldynl, 2022-03-08:
Thank you very much for detailed description.I am able to repeat it with the same result now.

foldynl, 2022-03-08:
Next release should repair the table.

commit 8fd785fb (2022-03-08): Fixed #11 - Missing Modes in Setting Dialog

under special conditions, it could happen that the Modes table was not filled correctly.
The fix adds missing records and improve the filling of this table.
closed 2022-03-08 by foldynl