foldynl/QLog #13: QLog 0.6.0, New-QSO window

DL2KI, 2022-03-07:
After logging a QSO the transmit power is taken from the tab "My Station" / "Power". Since for some bands (6m, 60m) the transmit power is limited, the content of the column "TX PWR" must be corrected in these cases. Or you have to change the field "Power" in the tab "My Station" before.

With CQRLog a manual correction of the transmit power is also necessary here.

You could leave it like this, introduce a band-dependent power setting, or take the transmit power from the rig.

DL2KI, 2022-03-07:
> TX PWR info is taken from the rig via HAMLIB. If you have connected your rig and your rig supports it then the field should be updated automatically. I will test it on my device, but it worked quite reliably (in case of ic7300).

- On my K3, I have set the power to 110W (maximum setting) and manually set the "Power" field in the "My Station" tab to 110.0W as well.
Now, when I slowly turn down the power control on the K3, the display in the "Power" field jumps to 10.0W when the power on the K3 is set at about 100W. The value 10.0W can no longer be set higher with the controller on the K3.

- If I delete the entry "[newcontact] / power=100" in QLog.conf and restart QLog, the value 100.0W is displayed in the field "My Station / Power". After "Connect Rig" the value remains at 100.0W, although the K3 is set to 110W.

- If I delete the entry "[newcontact] / power=100" in QLog.conf, and set the K3 to 50W, the value 100.0W is also displayed in the "My Station / Power" field when I restart QLog. After "Connect Rig" the power value jumps to 5.0W.

- If I do not delete the entry "[newcontact] / power=100" in QLog.conf, and set the K3 to 50W, the value 100.0W is displayed in the field "My Station / Power" when I restart QLog. After "Connect Rig" the display jumps to 5.0W.

- On the K3, I have 80m set to 110W, 60m set to 15W, and 160m set to 25W. When starting QLog, the K3 is set to 80m. In the field "My Station / Power" the value is 100,0W.
After "Connect Rig" the power value remains at 100.0W. 
After switching to 160m the value 2,5W is displayed.
After switching back to 80m the value 2,5W remains. Likewise after switching to 60m or other bands.

The transfer of the power from the rig does not seem to work reliably. However, in my tests I could not see what could be the reason for this.

I hope that my descriptions are understandable. Otherwise please ask specifically what I should try.

foldynl, 2022-03-08:
which rig model have you selected in the Setting/Hamlib tab? K3, K3S, KX3?

> * On my K3, I have set the power to 110W (maximum setting) and manually set the "Power" field in the "My Station" tab to 110.0W as well.
>       Now, when I slowly turn down the power control on the [...]

Please, could you try quoted scenario but you will start QLog in debug mode like this:

```QT_LOGGING_RULES=qlog.ui.newcontactwidget.function.parameters=true ./qlog```

it will generate debug messages related to newcontactwidget. Please, could you send me a result. I am mainly interested in the function "NewContactWidget::changePower"

DL2KI, 2022-03-08:
Here is the debug log. The power regulator on the K3 was turned down from 110W to 0W and then turned back up to 110W.

[debug.txt](https://github.com/foldynl/QLog/files/8208413/debug.txt)

[image]

DL2KI, 2022-03-08:
In "QLog.conf" I find the line "rig\model=229". However, in the new libhamlib4 the rig numbers have changed. For the Elecraft K3 now "2029" has to be entered.

foldynl, 2022-03-09:
Wolfgang,

it seems that you are using an old version of HAMLIB. Based on the debug message and screenshot, I can say:

1) Your Hamlib containts K3 driver version 20170812. I can see  in hamlib source code that this version was used between Aug 12 , 2017 and May 30, 2019

2) QLog reads the power information from Hamlib and your hamlib returned the information what you reported
```8:30:31.611 [DEBUG   ][qlog.ui.newcontactwidget.function.parameters]   => 3.3 [void NewContactWidget::changePower(double):ui/NewContactWidget.cpp:1240] ```

It means that hamlib returned 3.3W in your case.

3) Qlog.conf contains rig model 229 because it is a number used by Hamlib 3.x or older. Number 2029 is used by a newer hamlib (hamlib renumbered the rigs somewhere around Feb 18, 2020 - I guess that it was introduced in v4.0). To be honest, the guys from Hamlib do a good job, but quite often they change things in a pretty radical way. 

In your case, the solution is to upgrade hamlib (if it is possible). But I'm a little taken aback by the fact that you wrote that you use Kubuntu 21.10. I don't have installed it , but I use Ubuntu 21.10 and there is hamlib in version 4.0-7 (K3 driver in version 20201214.2)

```rigctl -l | grep K3```
```2029  Elecraft               K3                      20201214.2      Stable      RIG_MODEL_K3```
```2043  Elecraft               K3S                     20201214.2      Stable      RIG_MODEL_K3S```

DL2KI, 2022-03-09:
Hi,

my Hamlib is installed in version 4.0.
```
 ~ $ rigctl -V
       rigctl(d), Hamlib 4.0
 
 ~ $ rigctl -l | grep K3
      2029  Elecraft               K3                      20201214.2      Stable      RIG_MODEL_K3
      2043  Elecraft               K3S                     20201214.2      Stable      RIG_MODEL_K3S
```

Also installed is the "libhamlib2 3.3-10build1". Uninstalling this version is not possible without also uninstalling qlog.

Now I made some power settings on the K3 and checked the output via "rigctl".
```
# Elecraft K3
#
rigctl -m 2029 -r /dev/ttyS0 -s 38400

110W   l RFPOWER = 1.000000
100W               0.909091
 90W               0.818182
 50W               0.454545 
 10W               0.090909
  1W               0.009091
  0W               0.000000
```

After switching to "Kenwood TS-870S" the following output appears:
```
# Kenwood TS-870S
#
rigctl -m 2010 -r /dev/ttyS0 -s 38400

110W   l RFPOWER = 1.100000
100W               1.000000
 90W               0.900000
 50W               0.500000 
 10W               0.100000
  1W               0.010000
  0W               0.000000
```

So the problem seems to be within hamlib. The difference seems to be a constant 10%.

Entering a "Kenwood TS-870S" is possible in the "Settings / Hamlib" tab, but generates an error message in "Rig Connect".

[image]

OK, you can continue to observe the problem and wait and see if you get more feedback from other users later.

foldynl, 2022-03-09:
It seems to be an issue of the QLog Build Process. Currently, QLog Installation package is prepared only for Ubuntu LTS (Focal - 20.04) where the installation package depend on Hamlib package from Ubuntu 20.04 (where hamlib 3.3 is present).

**It is not your issue.** I have to change the build process and let you know when a "native" installation package for 21.10 will be available. I am planning to prepare Qlog fix release 0.6.5 this weekend.

foldynl, 2022-03-10:
I just released a new release that fixes most bugs (missing modes, Setting dialog in dark mode etc). Newly released for Ubuntu 21.10 and 22.04. This should fix the problems with hamlib. However, you will need to uninstall the current qlog and install the appropriate variant for your operating system - versions for 21.10 and 22.04 requires hamlib 4.x

DL2KI, 2022-03-11:
Hi,

thank you very much. I have uninstalled the version 0.6.0 and installed the package "qlog_0.6.5-1~impish_amd64.deb".

The modes are now complete and they are also taken over by the K3.

Elecraft uses a mode "DATA" for different submodes. A description can be found in the K3 manual , page 31, the KX3 manual, page 18 and in the K4 manual.

In the mode "DATA" the following outputs are displayed depending on the selection by the "DATA MD" key with "rigctl".
```
"AFSK A" 
  Rig command: m
  Mode: PKTLSB
  Passband: 400

"FSK D"
  Rig command: m
  Mode: RTTY
  Passband: 400

"PSK D"
  Rig command: m
  Mode: PKTUSB
  Passband: 400

"DATA A"
  Rig command: m
  Mode: PKTUSB
  Passband: 400
```

The transmitting power is not taken over correctly here either. After "Connect Rig", the power display in the "My Station" tab jumps to 10.0W.

"rigctl" also gives wrong results. The values are constantly 10% too low.
```
"Elecraft K3" (rigctl -m 2029)

K3 = 110W
Rig command: l RFPOWER
Level Value: 1.000000

K3 = 90W
Rig command: l RFPOWER
Level Value: 0.818182
```

But then the problem seems to be with "hamlib 4". With "Kenwood TS-870S" the outputs are correct.
```
"Kenwood TS-870S" (rigctl -m 2010)

Rig command: l RFPOWER
Level Value: 1.100000

Rig command: l RFPOWER
Level Value: 0.900000
```

foldynl, 2022-05-20:
I am closing the issue. If some of mentioned issue again occurs, please, open a new issue.

closed 2022-05-20 by foldynl