foldynl/QLog #17: Transfer of data from 'rigctl' to QLog

DL2KI, 2022-03-17:
Based on some tests (Discussions / Show and tell) I suspect that the data transfer from 'rigctl' partly does not work.

In the case of "Elecraft K2" this could be due to a too fast "poll rate". In the case of the "K3", the display "VFO A" in the "Rig" window remains after switching the VFO A/B.

For further tests I am gladly available (except 20.03 to 27.03).

foldynl, 2022-03-18:
I will prepare a "bugfix" branch where the pool rate will be set to 1s. I will send you a branch name later for your test. The bugfix branch will be derived from v0.6.5 therefore you will not have to worry about your setting or QSO database.

Please, be careful when using the "testing" branch. Things for the next release may appear in this branch, so databases and settings may change in this branch.

foldynl, 2022-03-18:
I have forgotten. 1s poll rate will be just for your test. If the change will work then I add a poll rate to Setting dialog.

foldynl, 2022-03-18:
Please, could you test the branch "bugfix-17"? There is the Polling interval set to 2s.

DL2KI, 2022-03-18:
Hi,

thanks for the bugfix test version.

The "polling rate" does not seem to be the cause, although the K2 needs a longer "polling rate".

The display still switches back and forth between PWR and frequency, as seen in the video. Here, only the switching rate is slower.

But after studying the "Elecraft KIO2 Programmers Reference" things are clearer now.

When executing the command "l RFPOWER" in "rigctl", the display of the K2 also changes to the PWR display for a short time. This is due to the fact that the power, which is queried via "PC (Power Output Level; GET/SET)", is read from the display (DS (Display Read; GET only)).

Please refer to the "Elecraft KIO2 Programmers Reference".
[]https://ftp.elecraft.com/K2/Manuals%20Downloads/KIO2%20Pgmrs%20Ref%20rev%20E.pdf

It is one of the peculiarities of the K2 that many of the K2 Get-commands do not return their results directly, but these are read from the display via the "DS" function. This makes it necessary to briefly show the power value in the display.

If this value is regularly queried by QLog, the alternating display of PSW / frequency is therefore OK. Therefore, the behavior will only change if there is no regular polling of the power level from the K2.

foldynl, 2022-03-18:
OK, thanks for your quick response. Based on your description, the poll interval is not an issue. Therefore I close the issue now. The rest issues with K2/K3 are solved in separate issues.

closed 2022-03-18 by foldynl