foldynl/QLog #29: Band Changer

DL2KI, 2022-04-08:
The Band Changer in the Rig Widget does not work for me. 
No band selection is offered.

foldynl, 2022-04-09:
Does it occur for your K2 or K3? Band Changer is filled from hamlib - it was quite reliable when I tested it

DL2KI, 2022-04-09:
Hi,

The effect occurs with both the K3 and the K2. The dropdown menu has no content.

[image]

foldynl, 2022-04-09:
I am sorry. You talked about the Band Changer. It is  not filled from from hamlib but from the profile setting. Do you have defined TX range in the Rig Profile? Setting-> Equipment-> Rig and for the profile field TX Range 0 - 30MHz? I wanted to fill it from hamlib but some rigs have  sub-variants and it is not easy to recognize which sub-variant you use. Therefore the operator has to define the range for particular rig manually. 
[image]

DL2KI, 2022-04-09:
Oh, sorry. Thanks for the info.

Now the band selection works. I had overlooked that in the wiki.

closed 2022-04-10 by foldynl