foldynl/QLog #32: Rig detection when pressing "Connect Rig"

DL2KI, 2022-04-09:
If you press "Connect Rig" with the transceiver is powered off, QLog GUI freezes. After about 24 seconds a window "QLog Warning" appears - "Rig Error: Open Connection Error - Communication timed out".

In "CQRLog" the switching of the two powered off rigs is possible without the GUI freezing. In the window "TRX Control" the frequency "0" is then displayed. The operation of the GUI is still possible.

Maybe an error handling can be added to this action to prevent the GUI from freezing.

foldynl, 2022-04-09:
how long do you have a Rig poll interval?

DL2KI, 2022-04-09:
Hi,

Rig poll interval is 500ms in both cases (K3, K2). So far it looks like the K2 can also handle this.

foldynl, 2022-04-10:
Cannot explain your 25 seconds. Timeout should occurred within 2 seconds but you are right, it is unacceptable GUI freezing during connection. Need to be fixed

DL2KI, 2022-04-10:
Hi,

I just tried this again several times with K2 off. When "Connect Rig" is selected, this message appears after about 25 seconds.
[image]

I also don't know where the big time difference comes from.

foldynl, 2022-04-10:
I can explain the 25 seconds. Hamlib does: sent command, detect timeout, if timeout then retry. 

For K2, there is timeout detection time 2seconds but retry limit is 10 (10 times to try the command and then hamlib reports an error to application). If I add some internal locks in QLog and a slow retrieving from rig(s) then the total time of "error detection" can take 25seconds - that's crazy.

It is the same issue, as you reported in the issue #33. 

QLog has to be prepared for log responses from rigs.

foldynl, 2022-04-22:
fixed in release 0.8.0

closed 2022-04-22 by foldynl