foldynl/QLog #35: ADIF-Import

DL2KI, 2022-04-10:
In "Discussions - Import of ADIF data from other programs " you wrote:

> Therefore I decides to implement following algorithm.
> 
> If QLog imports ADIF file with NAME field then it is saved to QLog's NAME_INTL field (without any conversion), because it can contain , incorrectly, UTF-8 characters from source application.
> 
> QLog tries to remap accents string to accent-less string and save it to NAME field. This field is marked as NAME (ASCII) and it is strictly used for ADI export.

When I save two test QSO's, the names and QTH's are saved in the NAME / NAME_INTL and QTH / QTH_INTL fields as described above.

See the first two log entries.

[image]

Then in QLog I deleted the existing log and re-imported the current ADI file from CQRLog. Now here you can see that the import of the fields NAME / NAME_INTL and QTH / QTH_INTL is not done as described above.

[image]

Most of the NAME_INTL and QTH_INTL fields are empty.

foldynl, 2022-04-10:
Please, could you share a ADIF record(s) (file) for the QSOs where the fields are missing?

DL2KI, 2022-04-10:
Hi,

here are the logbook excerpts:

```
cqrlog_adif-test.adi.zip:     Logbook extract from CQRLog
testlog.png:                  image of the records

testlog-export.adi.zip:       adi export from QLog
testlog-export.adx.zip:       adx export from QLog
testlog-export.json.zip:      json export from QLog
```
[cqrlog_adif-test.adi.zip](https://github.com/foldynl/QLog/files/8458652/cqrlog_adif-test.adi.zip)

[image]

[testlog-export.adi.zip](https://github.com/foldynl/QLog/files/8458656/testlog-export.adi.zip)
[testlog-export.adx.zip](https://github.com/foldynl/QLog/files/8458657/testlog-export.adx.zip)
[testlog-export.json.zip](https://github.com/foldynl/QLog/files/8458658/testlog-export.json.zip)

foldynl, 2022-04-10:
my apologies, my mistake. It seems that I did not "commit" the patch. I will fix it in the next release.

commit c17404a4 (2022-04-14): Fixed #35 - ADIF Import and converting to INTL fields
closed 2022-04-22 by foldynl