foldynl/QLog #40: New QSO, display of existing data from previous QSO's.

DL2KI, 2022-04-12:
If a call is entered in the 'Callsign' field that has already been worked, the previous QSO's appear in the QSO table.

In the 'New QSO' area of the main window some data (Name, QTH, Locator, Cont, ITU, CQ, State) appear to the callsign, which are obviously taken over from the last QSO, because the data are displayed even if the 'Callbook' is switched off. The name is taken from 'name', the QTH from 'qth'. It would probably be better here to display data from 'name_intl' and 'qth_intl', as these fields tend to be displayed as the default in the QSO table.

As far as I can see, the data of the fields 'comment_intl' and 'notes_intl' or 'darc_dok' are not taken from the last QSO.

But from my point of view this would make sense, because then you could reuse the contents completely or partially. Especially with 'Comment' and 'My Notes' it is often useful to be able to fall back on the information from the last QSO.

For the fields 'Age', 'E-Mail' and 'URL' it also seems to me to be useful, because these data remain relatively constant.

foldynl, 2022-04-12:
It is good idea.

1) you are right that name_intl, qth_intl etc should be used...I will fix it
2) comment_intl - I guess that it is not good idea to fill it from previous QSO because the field should be QSO-related...I won't add it
3) notes_intl - I will add it because it is operator-related notes
4) dark_doc - unfortunately, i am not familiar with DOK - I know what it is but I don't know exactly the rules. **Please, could you specified whether one callsign can have two or more DOKs during the time?** If it is possible to have multiple DOKs with one callsign then it is not good idea to fill DOK from previous QSO.
5) Age can vary - will not be filled from previous QSO
6) Email - will be filled from previous QSO
7) URL - will be filled from previous.

DL2KI, 2022-04-12:
Hi,

The DOK is the identification of a local association in the 'German Amateur Radio Club' (DARC). This identifies whether a radio amateur is a member of the DARC.

It is composed of the one-letter district identifier and a two-digit number. For some contests and for many diplomas DOKs are essential as counters. But collecting as many DOKs as possible on as many bands as possible as an end in itself is also very popular, analogous to collecting DX areas. To support the activities in the DOK area there are the DOK exchanges and the DOK best list.

Besides the regular DOK there are also special DOK, which are activated for certain occasions temporarily or for general purposes permanently.

Usually the radio amateur is a member of only one DOK (local association). Therefore, this value is largely constant and must be changed only at Beadrf. However, since there are also special DOK's for certain occasions, there may also be deviations here.

---

Yes, but only once a year. If one has several QSO's in the year with individual radio amateurs, nobody will ask with each QSO for the age. Basically you can do without the field.

I personally enter the age only if the QSO partner names this himself. The exact age is also not important for me. It is rather a 'guideline' for discussions with further QSO's, which I enter so far in "notes".

---

Comments on the last QSO can be useful to have a starting point for a conversation at a new QSO.
E.g. 'bad conds', 'vy QSB', 'QRP-QSO', 'QSO on vacation from Italy' etc.

> Hi, at the last QSO you were on vacation in Italy.

> Hi, At the last QSO your signal strength was very low. Today your signal is much better. Are you using a new antenna or a different rig?

One can enter this if necessary however also in "My Notes".

foldynl, 2022-04-12:
Many thanks for your response. getting back to DOK. 

> Usually the radio amateur is a member of only one DOK (local association).

Apologies for my stupid question but: does it mean when the operator changes the QTH (in Germany), the DOK remains unchanged?

> Comments on the last QSO can be useful to have a starting point for a conversation at a new QSO.
I guess that My_note should be used for personal notes. The Comment is used for sending a message to eQSL as a QSL Messages. On the other hand, there is no problem to add Comment as a column for filling from previous QSO. I will try to add it and we will see. Maybe it will be useful.

DL2KI, 2022-04-13:
Hi,

> Apologies for my stupid question but: does it mean when the operator changes the QTH (in Germany), the DOK remains unchanged?

There really aren't any stupid questions, hi.

There are about 1,000 local chapters in the DARC here. As a rule, a radio amateur becomes a member of a local association near his home, so that he can meet other members there in person.

If you move to another QTH, you can join the local association there (and change the DOK), or stay with the old local association. 

For me it is also the case that my 'local association' (K08) is about 60 km away, but I still go there because I know the people there. We also have our own club rooms in a public building with some permanently mounted antennas and can therefore also use technical equipment there..

---

> I guess that My_note should be used for personal notes. The Comment is used for sending a message to eQSL as a QSL Messages. 

You are right. You should do it the way you suggest. CQRLog also does not transfer the information to the new QSO window.

However, the 'My Notes' field in CQRLog is located directly next to the comment field so that you always have the contents directly in view. Obviously I have confused the two fields.

commit d6cbdece (2022-04-13): Fixed #40 - Added DOK field
commit 9ea477c1 (2022-04-13): Fixed #40 - New QSO, display of existing data from previous QSO
closed 2022-04-22 by foldynl