foldynl/QLog #50: "Connect Rig" when the rig is powered off

DL2KI, 2022-04-23:
If "Connect Rig" is selected when the rig is powered off, a warning message will appear after about 24s.

[image]

If "Connect Rig" is selected and then "Connect Rig" is switched off again after a few seconds, the user interface freezes. After about 20s the same warning message as above appears.

From my point of view it would be better to do without this technical warning message. You could program it internally so that there is simply no visible reaction and the checkbox "Connect Rig" is deactivated in the background.

If the software does not react after "Connect Rig", the user will search for the cause himself and find out that the rig is not powered on or the connection cable is not plugged in. At best, an informative, non-technical warning message could be displayed, e.g. "Cannot connect rig", or similar.

foldynl, 2022-04-25:
> If "Connect Rig" is selected when the rig is powered off, a warning message will appear after about 24s.

Unfortunately, I am not able to I am not able to affect this time. I tuned this in v0.8 and got responses around 2-3 seconds. We've talked about this before. K3 have long timeouts set to detect anything in Hamlib.

> If "Connect Rig" is selected and then "Connect Rig" is switched off again after a few seconds, the user interface freezes. After about 20s the same warning message as above appears.

I will try to simulate but as I wrote above. In my case, an error detection is 2-3 seconds. But you are right, GUI must not freeze in this case. Are you using v0.8? I tried to fix GUI freezing  in 0.8.

> From my point of view it would be better to do without this technical warning message. You could program it internally so that there is simply no visible reaction and the checkbox "Connect Rig" is deactivated in the background.

I am not sure if I understand you correctly. you want to say that QLog should suppress the errors from Hamlib?

> If the software does not react after "Connect Rig", the user will search for the cause himself and find out that the rig is not powered on or the connection cable is not plugged in. At best, an informative, non-technical warning message could be displayed, e.g. "Cannot connect rig", or similar.

**You are right. The error should be displayed for non-programmer user.**

DL2KI, 2022-04-25:
> Unfortunately, I am not able to I am not able to affect this time. I tuned this in v0.8 and got responses around 2-3 seconds. We've talked about this before. K3 have long timeouts set to detect anything in Hamlib.

Yes, then there is probably nothing you can do about it. Unless you find a way to monitor the connection to the rig / rotor outside Hamlib, when there is no way inside Hamlib to monitor the connections independently of the rig / rotor.

> I am not sure if I understand you correctly. you want to say that QLog should suppress the errors from Hamlib?

Yes, one possibility is to suppress these error messages in general and display the error in another way.

CQRLog: No error message, frequency display jumps to "0".
QLog: No error message, rig profile color changes from green to red.

You have already suggested another solution. Intercept error messages and display them in a form that is more readable for "normal users".

I think that both variants are usable. However, displaying human readable error messages is more noticeable to the user and therefore perhaps a bit better.

Again, these are problems that do not directly affect the function of QLog. The software looks very good so far. Maybe other ideas will arise later.

DL2KI, 2022-04-27:
Hi,

I have experimented a bit with rigctl and the "-C" option.

The rigs K2 and K3 are both switched off. Input on the command line.

**K2**: 

**`rigctl -m 2021 -r /dev/ttyUSB0 -s 4800 -C stop_bits=2`**
The error message 'rig_open: error = Communication timed out ...' appears after about 23 seconds.

**`rigctl -m 2021 -C timeout=100 -C retry=1 -r /dev/ttyUSB0 -s 4800 -C stop_bits=2`**
The error message 'rig_open: error = Communication timed out ...' appears after < 1 second.

**K3**: 

**`rigctl -m 2029 -r /dev/ttyS0 f -s 38400 -C stop_bits=1`**
The error message 'rig_open: error = Communication timed out ...' appears after approx. 6 seconds.

**`rigctl -v -m 2029 -C timeout=100 -C retry=1 -r /dev/ttyS0 -s 38400 -C stop_bits=1`**
The error message 'rig_open: error = Communication timed out ...' appears after < 1 second.

Maybe this is an approach to check the connection to the rig/rotor faster and independent of the rig/rotor, at least for "Connect Rig". If the connection is successful, it should be reestablished immediately without the parameters 'timeout' and 'retry'.

Whether this is useful or necessary is another question, because the timeouts do not affect the basic function of QLog.

foldynl, 2022-04-27:
That's true QLog can modify Hamlib's rig setting but how should be done?

I trust that the default parameters are selected as the best settings. Unfortunately, I can't afford to let users set up various internal parameters, because that would be confusing and unsupported from QLog site - many user have many different sent is nightmare of all support teams.

I'll try to come up with something, but I definitely don't want to go the way of expanding the Setting Dialog.

DL2KI, 2022-04-27:
> but I definitely don't want to go the way of expanding the Setting Dialog.

Yes, I agree with that. 

The problem (Connect Rig when Rig is switched off or not connected to PC) does not affect the function of QLog. I just wanted to try out for myself whether you can detect a switched off rig faster with Hamlib.

A permanent reduction of the preset timeout times for the rigs / rotors is of course not useful.

The probability that '"Connect Rig" when the rig is powered off' or 'Disconnecting the rig' occur are rather theoretical. From my point of view the two issues #50 and #51 can be closed.

foldynl, 2022-04-29:
I don't think than there is nothing to do. I see following issue:

1) GUI must not freeze when an operator presses "disconnect"
2) Hamlib error messages must be more "user-friedly"

closed 2022-05-20 by foldynl