foldynl/QLog #51: Disconnecting the rig

DL2KI, 2022-04-23:
If the rig is switched off while the QLog is running, this warning message appears after approx. 46s.

[image]

And the color of the rig profile in the "Rig Widget" changes from green to red.

If the rig is switched off while the QLog is running, and then the "Connect Rig" checkbox is manually deactivated, the GUI freezes. After about 46s the same warning message as above will appear, and  the color of the rig profile in the "Rig Widget" changes from green to red.

Again, it seems to me that internal "transparent" error handling without a technical warning message would make more sense. At best, one could display an informative, non-technical warning message. e.g. "Connection to rig interrupted", or similar.

Also, the time duration for the detection of these states seems to me to be too long.

DL2KI, 2022-04-24:
[issues/50](https://github.com/foldynl/QLog/issues/50), [issues/51](https://github.com/foldynl/QLog/issues/51)

Hi, 

some thoughts about the problem.

Is there a watchdog routine in Hamlib for these purposes? Is it possible to include such a monitoring in QLog itself? The rig is always in view during radio operation. But not a rotor. What happens if the rotor blocks during operation?

Similar problems could also occur when operating a station in remote mode. How is this handled in remote servers?

foldynl, 2022-04-25:
You have good questions but I cannot answer them now. Hamlib is little bit magic for me. For example. When you call open for existing port and rig is off then Hamlib returns OK for rig_open command but then Hamlib returns the timeout error for get_Freq. This caused your problems. 

> Also, the time duration for the detection of these states seems to me to be too long.

Unfortunately, timeouts are defined per rig. it is not under my control. 

> Again, it seems to me that internal "transparent" error handling without a technical warning message would make more sense. At best, one could display an informative, non-technical warning message. e.g. "Connection to rig interrupted", or similar.

At this moment, the main message from this issue is to improved the error message for non-technical users.

> f the rig is switched off while the QLog is running, and then the "Connect Rig" checkbox is manually deactivated, the GUI freezes

It was tuned in 0.8..Are you using v0.8?

DL2KI, 2022-04-25:
Hi, 

sorry for the unclear description of the test situation. For the tests I used version 0.8 and the K2.

> You have good questions but I cannot answer them now. Hamlib is little bit magic for me. For example. When you call open for existing port and rig is off then Hamlib returns OK for rig_open command but then Hamlib returns the timeout error for get_Freq. This caused your problems.

Yes, I also had a look into the Hamlib code, and I can also understand one or the other. But all in all, as a hobby programmer, I lack the professional view.

But I think I read something in a changelog about a "watchdog" for the rotor control. Therefore I had addressed this.

These are problems that do not directly affect the function of QLog. It can be that later ideas for an improvement are given.

> Unfortunately, I am not able to I am not able to affect this time. I tuned this in v0.8 and got responses around 2-3 seconds. We've talked about this before. K3 have long timeouts set to detect anything in Hamlib.

Just now I compared it again with CQRLog and both rigs. 

CQRLog:

If the connected K3 is switched off, the frequency display switches to "0" after about 6 seconds without an error message appearing. With the K2 it takes about 12 seconds.

QLog:

If the connected K3 is switched off, the error message appears after approx. 12 seconds. With the K2 it takes approx. 46 seconds.

Yes, there will probably be nothing you can do about it then. Unless you find a way to monitor the connection to the rig / rotor outside of Hamlib.

> At this moment, the main message from this issue is to improved the error message for non-technical users.

This is a good idea to catch all internal errors and display error messages understandable for "normal users".

foldynl, 2022-04-29:
issue is solving by #50