foldynl/QLog #53: Rig/Rotor - Connect/Disconnect

DL2KI, 2022-04-29:
Since the rig/rotator is usually connected after starting QLog, it might be easier for operation to move the checkboxes from the "Equipment" menu to the 'Rig' / 'Rotator' widgets, or to place an additional checkbox or buttons here.

Example:

[image]

[image]

foldynl, 2022-05-01:
I thought about whether to do it like you wrote. But in the end, I stayed with the solution that is now. And I'll tell you why. This is because I would like to keep the GUI as clear as possible. Connect / Disconnect is used very rarely, so I didn't want to add another button that won't actually be used.

When QLog becomes "more stable", then the shortcuts will be defined and I guess that connect is a hot candidate for it.

At this moment, I have no plan to implement the buttons for connect/disconnect as you show above.

foldynl, 2022-05-12:
Basic shortcuts will be introduced in v0.9

foldynl, 2022-05-20:
basic shortcuts via ALT + menu option in v0.9.0

closed 2022-05-20 by foldynl