foldynl/QLog #55: Compile QLog under Debian 'bullseye' 32bit

DL2KI, 2022-04-30:
Hi,

I tried to compile QLog on an older IBM-T60 laptop. The laptop is running 'Debian bullseye 32bit'.

After starting QLog I get this message:

> t60@lenovo-t60 ~/Software/QLog/QLog $ ./qlog
> Hash collision!!! Fatal error!!
> 08:13:04.720 [WARNING ][             ]  => QObject::killTimer: Timers cannot be stopped from another thread [(null):(null):0] 
> 08:13:04.720 [WARNING ][             ]  => QObject::~QObject: Timers cannot be stopped from another thread [(null):(null):0] 
> 08:13:04.720 [WARNING ][             ]  => QObject::killTimer: Timers cannot be stopped from another thread [(null):(null):0] 
> 08:13:04.720 [WARNING ][             ]  => QObject::~QObject: Timers cannot be stopped from another thread [(null):(null):0] 
> t60@lenovo-t60 ~/Software/QLog/QLog $ 
>

Where could I start troubleshooting?

Here is the compiler log and the debug log.

[_compiler.zip](https://github.com/foldynl/QLog/files/8596338/_compiler.zip)
[_debug.zip](https://github.com/foldynl/QLog/files/8596339/_debug.zip)

foldynl, 2022-05-01:
Thanks for reporting the issue.

"Hash collision!!! Fatal error!!" is a message from hamlib that an unknown Rig ID is used.

Unfortunately, the Hamlib project is very dynamic and the behavior of the library varies from version to version. This is the issue what you reported. When QLog starts and does not have a defined Rig profile, it uses the default profile Rig ID 0. This works with higher versions but it seems that Debian bullseye Hamlib does not support it. 

Unfortunately, I don't have a quick solution for you. It is necessary to fix it in the source code of QLog. 

I must say that I am a little disappointed with the behavior of hamlib - especially version to version changes. But there is nothing to be done because there is nothing else to use here. Or am i wrong?

DL2KI, 2022-05-01:
Hi,

I have copied the QLog database once to the T60. In fact, QLog now starts without error message.

Your hint was correct. Thanks a lot.

Unfortunately the screen resolution of 1400x1050 is too low, so I can't use QLog here. But that's not a problem, since I don't use the laptop for this so far. If this would be useful for portable operation, you can use CQRLog.

As for hamlib, unfortunately I am only a user. Therefore I am not so familiar with the problems related to it.

If you choose to use it, you probably have to accept it that way, hi. 

As far as I know Fldigi uses a specially developed for Fldigi 'RigCAT xcvr control'. In the Fldigi manual you can find hints about it. The sources should be available.

If it is worth to use RigCat is another question. In the end you will have to consider the effort that is required for both systems.

closed 2022-05-20 by foldynl