foldynl/QLog #59: testing_0.9 - User-defined Spot Alerts

DL2KI, 2022-05-11:
Spot Alerts' is a feature in conjunction with the DX-Cluster widget. However, the 'Alert Rules' are set via the 'Logbook' menu. In this menu, 'Alert Rules' is the only menu item that is not directly related to the logbook itself.

Perhaps the 'Alert Rules' menu item would be better placed in the Spot Alerts icon drop-down menu, or elsewhere.

foldynl, 2022-05-11:
you're right. It's a good observation. I hesitated where to place it. In the end, I decided to put it in the main menu. But your solution is better. Many thanks for your comment, I'll change it. 

Just be careful when using these "testing_x" branches. I can extend the database between commits. And then it depends a lot on what commit you used. It can lead to the corruption the database. 

There are only two ways to fix such damage database
1) deleting the database
2) manually applying later SQL statements - which is not very convenient.

I have to keep an eye on this when it develops on my computer.

for this purpose, I am planning to change branches for upcoming releases.

DL2KI, 2022-05-11:
Hi,

i start the testing-releases with a bash-script, which saves the databank temporarily and restores it after the program ends. This works quite well so far.

DL2KI, 2022-05-11:
I have noticed a few things with the 'Spot Alerts' so far, but I still have to deal with them in more detail. Since the function is not yet commented, it may also be that you still want to make changes there.

Here two things in advance.

If under 'Alert Rules' DX Callsign = 'I*'  or 'DL*' is set, also callsigns appear in the alert list which do not correspond to the selection.

DX Callsign = 'I*'
[image]

DX Callsign = 'DL*'
[image]

Then I noticed that the alert list fills up very fast and becomes confusing. Just after a few minutes I have more than 500 alers in the list (DX Calsign = 'D*').

From my point of view, the number of alerts should be reduced. But I still don't have an idea how to do that.

You could e.g. buffer duplicate alerts and show the most recent or last alert after a certain time or number of alerts. With this one could exclude the multiple alerts that occur in a short time interval.

In addition, the number of alerts in the alert list could be limited. Depending on time or on the number of entries in the list.

In the meantime I have reached more than 1400 alerts.

But, as I said, I still need to think about what would be practical.

foldynl, 2022-05-11:
I have prepared updated Wiki pages about it but they are not published yet. 

Briefly, DX Callsign and Comment are special fields. There **MUST** be a valid Regular Expression based on Perl-like regular expression syntax.

RegExp is powerful tool but it has to be precisely defined. Example rule `D.*` matches Callsign DL1AAS but also SV1DAY. If you want to match only DL stations then it is needed to used the regexp `^DL.*`

A good starting point is page https://regex101.com/ where you can fine-tune your expressions.

You are right, the duplicates are a problem especially when you are connected to RBN but as you can see the RBN network produces non-unique comments. So the question is, when you will somehow aggregate spots, you will ignore the comment ? I guess it is not possible to do it because Alert Rules can contain a regexp for comment. Therefore you may also be interested in what is written in the comment.

DL2KI, 2022-05-11:
Yes, thank you very much for the hint. I had already thought so. It is better to wait with such comments until there is documentation for the function.

> You are right, the duplicates are a problem especially when you are connected to RBN but as you can see the RBN network produces non-unique comments. So the question is, when you will somehow [...]

The large number of alerts in the alert list is rather theoretical. As a rule, the number of alerts will be reduced to a few with the appropriate specifications.

It depends on which goals the respective user wants to achieve with the alert function. 

I personally use in CQRLog a list with some callsigns with which I have frequent contact, so that I can address the OM's then if necessary. So it is enough for me to get a message about callsign and frequency. 

Other users may expect additional information.

RBN is of course generally interesting for CW operators. Therefore, you could think of a way to reduce the number of multiple spots that are close together in time. This certainly depends on how the alerts are processed internally in the program. Whether you limit the size of the alert list may also depend on the internal programming. But you can judge that better.

You can wait and see if other users with different requirements will contact you about this function later. Then you can still make changes here.

foldynl, 2022-05-13:
I postponed release 0.9 due to missing spot aggregation. I don't like the way it works now - you are right that alerting is just another window where the spots are showen. I'm thinking about how to do it.

DL2KI, 2022-05-13:
Yes, some features sound good in theory, but are often difficult to implement in a practical way.

I myself sometimes need some time to determine whether I like a functionality or not. Some things you have to use for a while to see the advantages and disadvantages.

The implementation in CQRLog with an external program (here: /usr/bin/notify-send) is not optimal either. With many alerts popup windows appear all the time. This disturbs more than it helps.

From my point of view this feature is only practical if you use settings that guarantee a small number of alerts that you will use. This is then the responsibility of the user. 

One should perhaps point this out in the Wiki and give some example settings with low alert occurrence. The user will then get a feeling for what is practicable for him.

foldynl, 2022-05-13:
Try to compile a branch "alert_aggregation". There is my attempt to aggregate spots in some way. Is it clearer now? I think so. 

Otherwise, I tried to sort it by time and it's not  readable - especially when you are connected to RBN. Despite the fact that spots are aggregated, their order changes very quickly

/usr/bin/notify-send cannot be used because 

1) spots can flow the notify system
2) QLog can run under Win (maybe MacOS). And these system have no notify-send binary.

You can use the Network Notification messages to process (show) the alerts externally.

DL2KI, 2022-05-14:
Hi,

i tried the 'alert_aggregation' branch and it seems to work much better. From my point of view, I would leave the functionality as it is for once. You have to work with it for some time and see how it behaves.

However, I need to get more involved with Perl-like regular expression to try out the possibilities. But some things work already. 
But it may be that many users of the alert function will have problems with it.

From version 0.9 I will import my current ADIF data from CQRLog into QLog and then use QLog primarily. One gets then a better impression about the operation of the software.

In 'Log4OM2' there is still a short acoustic beep-sigal for each alert. Personally, this would bother me for many alerts, others might find it useful. At least the acoustic signal must be able to be switched off.

Anyway, vy good work from you!

commit f93467ca (2022-05-11): Fixed #59 - Alert menu changes

- Alert Menu removed from Logbook menu
- Changed tooltip for regular expression fields - Perl-like syntax
commit 872e3138 (2022-05-15): Fixed #59 - Alert Spot Aggregation
closed 2022-05-20 by foldynl