foldynl/QLog #64: QLog database QSO sorting

DL2KI, 2022-05-20:
When migrating the ADIF data from CQRLog to QLog, I noticed that the records in the QLog database are stored sorted in reverse. After importing the data, the newest QSO is found in row 1 when the database table is opened with 'SQLiteStudio'.

If you now save a new QSO in QLog, this will be appended to the end of the database table. Therefore, when editing the latest QSO, you always have to scroll to the end of the database table.

Perhaps it would be clearer, if the newest QSO is always inserted at the beginning of the table when saving in QLog. Thus the table would be sorted then after descending date / time.

foldynl, 2022-05-22:
I am not sure if I understand you correctly but the order of QSOs in the database is determined by the order in which they were imported or entered. If you want to change the order of QSOs in "SQLiteStudio", you need to do this via sql "ORDER BY" and not via changing the internal logic in QLog

DL2KI, 2022-05-22:
Yes, this is not a problem.

I had only noticed after the import from CQRLog that the sorting in the database is reversed as in CQRLog. 

This leads to the fact that the data after the import from CQRLog are sorted by 'start_time' descending. The newest QSO can be seen in line id=1.

If now new QSO's are inserted with QLog, these are inserted however at the end of the table.

This creates a table that is not sorted uniformly. This affects the database table and the adx backup file.

You might be able to get around this by first sorting the adi file from CQRLog by <QSO_DATE:8> and <TIME_ON:4> in reverse order before importing. I will try this for my data.

For the use of QLog it doesn't matter. But if you want to use the adx-BackupFile in an external program or script, you have to consider that the data in the adx-file is not sorted uniformly.

Maybe it is possible that this is considered in the import routine of QLog and the imported data sets are adapted to the sorting of QLog.

foldynl, 2022-05-22:
If QLog imports QSOs then they are important "as-is". If QLog exports QSOs then they are (should be) exported order by START_TIME ascending (regardless of the exported format).

the structure in the database is only an internal thing of QLog and will not be modified now. The CQRLog has own structure which certainly makes a sense to CQRlog's developers.

However, if any issue is reported in the QLog export side, then this needs to be fixed/improved.

closed 2022-06-06 by foldynl