foldynl/QLog #65: Clublog- eQSL Service

DL2KI, 2022-05-20:
At the first call of the Service -> Clublog window the dropdown menu 'My GridSquare' does not show the locator from the set profile, but another one from another profile. With eQSL this also seems to be the case.
            
The upload is apparently done separately by profiles and not together for all QSO's.

Here I would have to still check, which consequences this has, if in the profiles different callsigns are indicated. e.g. IN3/DL2KI or DL2KI/P. Do these have to be registered in Clublog or eQSL before?

With the exact Funktionaliät with the Upload and Download I must concern myself however still more near. Therefore it can be also, that a thinking error is present with me.

foldynl, 2022-05-22:
Wolfgang, "My GridSquare" in Service Menu "eQSL, QRZ..." is taken from QSO database. It means that QLog shows only "My Squares" that are currently present in QSO database. If you do not have any QSO for the given grid yet, then this grid will not be displayed. In other words, grids are not taken from the profile but from the worked QSOs.

DL2KI, 2022-05-22:
Ok, thanks for the hint.

I'll have to look into this a bit more so that I understand the mechanism behind these functions in more detail.

closed 2022-05-22 by foldynl