foldynl/QLog #66: Subsequent QSO entry (Offline-Mode)

DL2KI, 2022-05-21:
The subsequent entry of QSO's from a paper log (portable) is cumbersome.

Date and time cannot be entered directly into the data fields. The input via a calendar widget or a ComboBox is considerably slower and error-prone.

If the corsor is in the 'Date' data field, a warning message 'Delete the selected contacts?' appears when pressing the DEL key. The functionality of the buttons next to 'Time On' and 'Time Off' is not described.

[image]

From my point of view, it would be easier if the date and time were displayed in simple text fields, and you could simply enter the values by hand in offline mode. The existing 3 buttons to the right of the times could be omitted. A button or checkbox 'Offline mode' would have to be added.

foldynl, 2022-05-22:
Using text field has several disadvantages. I will say only the most important. You have to validate an input data and the input data, especially time and date, has many formats around the world. Therefore the specific Date/Time GUI elements are used that have implemented all these validates. 

"Offline mode" is not necessary to add because you can stop the timer via shortcut "F8" or "Stop button". Offline mode is not clean solution. 

I tried to improve time/date fields

foldynl, 2022-05-22:
I have found a solution. To enable field focusing solves the issue. A fix will be a part of next release.

foldynl, 2022-06-05:
improved in v0.10

closed 2022-06-05 by foldynl