foldynl/QLog #67: Field contents in capital letters

DL2KI, 2022-05-21:
In the QLog database the contents of the fields ''my_city', 'my_city_intl, 'operator' are stored in capital letters for newly entered QSOs.

foldynl, 2022-05-22:
Yes, you are right. I am gradually adding information. I'm  going from more important to less important. It is not possible to solve everything at once. Therefore, I decided to leave empty some "non-important" fields yet and focus on making QLog usable for normal usage.

Otherwise, the mentioned fields ('my_cq_zone' and 'my_itu_zone' ... ) are not mostly used by external services (QRZ, eQSL), therefore it is not necessary to fill them at this moment. The external services ignore them or fill them in automatically. But as I wrote, you are right, it is need to fill them, If you know them.

To be honest, do you need them? my_lon, my_lat are duplicated etc. if we want to fill in all of them, I will have to ask the operator a lot of things during the initial setup and the next time the Station profile was added. I'm not sure what's worse - having some blank items in database or filling out long forms to set up your station properly.

I will try to add those information that I am able to deduce from the current information, or add some more.

DL2KI, 2022-05-22:
Yes, I can agree with that. I myself do not need this '_my' either. 

> I'm going from more important to less important. It is not possible to solve everything at once.

Nevertheless I am sometimes amazed about your programming progress. I hope that your motivation remains so high, hi.

foldynl, 2022-05-22:
> I hope that your motivation remains so high, hi.

I hope so, but it is more difficult. I'm getting to areas that I haven't explored much anymore - hamlib's real split mode detection is a bit of magic for me, some internal parts of hamlib are not well-documented, sometime there are many side-effects (like your k2 and k3). Hamlib developers also do a good job, but their changes between versions are sometimes very radical.

And last but not least, keep everything at the level of everything is running on Linux and Windows (when I don't really know how Windows works, because I am primary a Linux/Unix developer).

DL2KI, 2022-05-22:
Yes, I can imagine that you quickly get into difficult areas here, especially when you first have to get into the details.

If QLog becomes better known, you may find other programmers who can assist you more in your work. At some point it is hardly possible to do it alone.

Later there will be user support as well. This can probably not be done via Github, because that is probably not an option for the normal user.

I have now found a solution for me to convert the wiki pages into a PDF and ODT, so that I can then make my own notes in the pages.  But also here you will have to offer an 'online manual' to the normal user later on. Maybe in the style of the online help of CQRLog or with an online wiki, e.g. https://www.dokuwiki.org/dokuwiki

Perhaps after version 1.0, development will have to be reduced for a while in favor of this 'side work'. 

All in all, the effort to maintain such a software will not decrease. Therefore, I hope that you find your own way to keep everything manageable.

commit 78867bac (2022-05-26): Fixed #67 - Remove calls toUpper for ADIF fields where it is not needed (2)
commit a5c782d0 (2022-06-05): Fixed #67 - Remove calls toUpper for ADIF fields where it is not needed
closed 2022-06-05 by foldynl