foldynl/QLog #68: Viewing and editing field contents of existing QSO's

DL2KI, 2022-05-21:
Viewing and editing field contents of existing QSO's is very cumbersome if the columns are not displayed in the QSO table. For viewing or editing the desired columns have to be shown first and later hidden again.

In the meantime I only do this with 'DB Browser for SQLite' or 'SQLite Studio', because there I have an overview of all table columns for the QSO and I can easily change or add the contents, e.g. in 'notes' and 'notes_intl'.

A reasonable solution would be from my point of view:

- Right click in a row of the QSO table.
- Context menu 'Edit QSO'
- Transfer the QSO data into the data area below the table
- Fix the data contents, date, time etc.
- Change the font color to red, blue or green as a visible edit mode indicator.
- View or edit individual field contents by the user.
- Press Save button and apply changed QSO data.
- Change the font color to the original color.

foldynl, 2022-05-22:
I know about it. Editing is not a strong part of QLog, but there are many things to think about.

1) ADIF format define around 150 fields for QSO. I cannot show/edit all parameters "data" area below the table. It is impossible to show them in the one dialog.
2) OK, we can define a limited set of field that can be defined via "data" area. Then we will have another problem to define "the set" of fields. The set is different for HF, VHF, EME, MS stations. Each of these stations has its own settings, each operator has its own habit and necessities. It is also not possible to leave it to the operator which fields will be displayed in the edit window, as this would be crazy to set it.

 I'll tell you how I do it. I have set the visibility of all fields in QLog (Context menu -> Column Visibility). Then I move the important columns for me in the window on the front position (drug-drop). Then I  edit everything via Qlog - no external problem is needed.

I'm not defending myself, change the edit fields through some dialog. But it is necessary to keep in mind that there are 150 possible QSO fields and editing must be done clearly and without the need to set anything complicated.

DL2KI, 2022-05-22:
Yes, I can understand that, and it is not practical to implement because of the amount of possible fields.

Your solution is one of the ways to handle this. Normally the data fields are not edited very often.

If one assumes that the user uses the data fields, or parts of them, which are offered in the main window below the QSO table and in the tabs, one could limit oneself with an edit function to exactly these fields, which one can also edit before saving.

With me it is so that I add afterwards perhaps once 'My Notes', or or perhaps 'Age' or 'E-Mail'. That's not much.

It can also happen that you have to edit the data of the tab 'My Station', if you forgot to enter details like the name of the station profile or the used antenna before saving.

I understand that this is a matter that can be seen very differently. Maybe you should also wait here until opinions of other users arrive. It is simply so that one must generally use software first of all some time, in order to be able to form an opinion over the handling.

foldynl, 2022-06-26:
QLog v0.11 contains this enhancement.
Closing the issue report.
More info - https://github.com/foldynl/QLog/wiki/QSO-Detail

closed 2022-06-26 by foldynl