foldynl/QLog #77: - [NEW] Bandmap RX Mark Center (issue #69)

DL2KI, 2022-06-08:
Today I experimented a bit with this feature. This resulted in the following:

If you want to be active in a certain band segment, you set the bandmap to this range. E.g. to the upper CW part of the 80m band, 3550 kHz to 3570 kHz. Here there are more local QSO's. 

With the zoom function you can set the desired area accordingly well over the whole window height. Thus the band section can be observed well.

Example: Rig VFO on 3551 kHz. Bandmap is set to 3550 kHz at the top to observe the band section 3550 kHz to 3570 kHz.

Unfortunately the bandmap with the RX cursor centers after some seconds, so that an observation of an individually set band section is not possible.

If you click now on a callsign, which is e.g. in the upper or lower range of the bandmap, the data are taken over into the QSO window and the rig is adjusted accordingly. 

Unfortunately the bandmap (RX cursor) is now centered on the frequency of the selected callsign, so that the previously selected observation range is lost again.

You can try this out for yourself with this mental specification to get a feeling for it.

From my point of view the following should be possible:

- Setting the bandmap to a certain band section without automatic change of this section by the program, even if the RX frequency set on the rig is outside this set section. (Then no RX cursor is displayed). This is necessary to be able to observe certain band sections permanently.

- No automatic change of the set (displayed) band section, if a callsign within the section is clicked. Only the RX cursor should be set to the clicked frequency.

One would have to think about how to combine these requirements with https://github.com/foldynl/QLog/issues/69. I think that a solution can be found based on what is convenient for the user of the bandmap.

I hope that I have described my thoughts understandably.

foldynl, 2022-06-09:
yes, you are right...TX should be used. Thanks for your notice.

foldynl, 2022-06-09:
My apologies but can you write more info? I have reviewed the code that RX freq is really centered.

DL2KI, 2022-06-09:
Today I experimented a little with this function. The result is the following:

1: ---

If you want to be active in a certain segment of a band (e.g. contest-free area of a band, or some other area), you adjust the bandmap accordingly by moving the frequency band and using the zoom function, so that you can observe this area.

Example: Rig VFO at 3550.00 kHz. Bandmap is set at the upper edge to approx. 3545.00 kHz to observe the band section 3550 kHz to 3570 kHz.

[image]

After about 13 seconds, the bandmap centers with the RX cursor, so that observation of an individually set band section is not possible.

[image]

2: ---

If you now click on a callsign (e.g. '7L2KOZ/1/QRP', which is e.g. in the upper or lower part of the bandmap, the data will be taken over into the QSO window and the rig will be adjusted accordingly. 

[image]

But at the same time the bandmap (RX cursor) is centered on the frequency of the selected callsign, so that the previously selected observation range is lost again.

[image]

From my point of view the following should be possible:

- Setting the bandmap to a certain band section without automatic change of this section by the program, even if the RX frequency set on the rig is outside this set section. (No RX cursor is then displayed). This is necessary to be able to observe certain band sections permanently.

- No automatic change of the set (displayed) band section when a callsign within the range is clicked. Only the RX cursor should be set to the clicked frequency.

One would have to consider how to combine these requirements with #69. I think a solution can be found that accommodates the user of the bandmap.

One possibility would be a lock function of the bandmap, e.g. with a "lock symbol" next to the zoom buttons 'In', 'Out'.
[image]

By clicking on the symbol, the frequency display of the bandmap for the selected band would be if a CAT connection exists.

Another option would be a 'CAT on/off' button to enable/disable the bandmap connection to the transceiver (when transceiver control is started).

There are certainly other solutions.

As I have read in the help texts of other log programs (RumLog, Log4OM2, ...) there are different approaches here. So obviously you have to make the details dependent on how the bandmap design is going to be determined in the future.

I hope that I have described my thoughts in an understandable way. One has to see how this looks in the result. It would be good if some other OM's could give their opinion.

DL2KI, 2022-06-09:
A CAT on/off button is of course not such a good idea in an active bandmap!

foldynl, 2022-06-10:
As far as I know, no one except you actively contributes to QLog. What a pity. So it's up to us how QLog will look like, hi.

You have described several ideas. I'm not sure whether I understood all of them - I have to analyze it properly. BTW, Do you know that  "Center RX" function can be switched off via Bandmap's context menu? It means that if you need to view the bandmap without centering, this can be used. 

Otherwise, during centering and no activity, the  bandmap is redrawn every 30 seconds (or minute), because it is necessary to perform "automatic spot clearing".

DL2KI, 2022-06-10:
Yes, sometimes longer thinking and trying is better, as you often need to observe the behavior for a long time first, and see how the practical handling is.

Therefore, you do not have to implement all thoughts directly.

> Do you know that "Center RX" function can be switched off via Bandmap's context menu? It means that if you need to view the bandmap without centering, this can be used.

Thank you for the hint.

No, I hadn't seen that yet. Context menus don't stand out directly, so it's often overlooked here. But I will try it out. Maybe that's already a solution.

dl2ki, 2023-04-17:
I think you can close the issue. 
As it is now, you can work very well with the bandmap.

closed 2023-04-18 by foldynl