foldynl/QLog #85: Autobackup

DL2KI, 2022-06-17:
> Wiki: QLog automatically exports the full log to the ADX file before each run.

I start QLog via a BASH script, and create the entries for an online log from the current backup file after exiting QLog.

If I start QLog to add QSO's from my paper log, these QSO's are missing after quitting, because QLog creates the backup file at startup.

Is there a possibility to create the backup file in general after quitting QLog? From my point of view, in both cases it is ensured that the current data is saved.

foldynl, 2022-06-23:
my intention was that the backup should be done before each run because

1) the backup is performed just before a possible migration to a higher version of the database - between QLog versions, if it is necessary, the internal structure of the database is automatically updated.
2) Not all application terminations are expected (application crash). And in these cases, no backup is made.

That's why I think the pre-start backup is safer. what do you think?

I have a note to the internal structure of the database. Beware that it may change. If you use own utilities to access the DB that's great but I cannot guarantee that the internal structures will not change. For example, the prepared version 0.11 modifies the contacts table due to the change of the ADIF version from 3.1.2 to 3.1.3 (ADIF added 3 new fields).

DL2KI, 2022-06-23:
Hi,

your thoughts about the timing of the backup seem to make sense to me. For the normal user of QLog it makes no difference. 

However, if someone wants to continue processing the current ADIF data from QLog after quitting the program, this is a special case. The person will then surely be able to find a solution for this himself.

I have changed my application so that I first read the table 'contacts' with 'sqlite3' from the database in 'csv format' and then process the data further. Since the first row contains the field names, these are always completely available internally. 

I then have to take adjustments into account in the application itself, if necessary. When importing the ADIF backup file, however, this is also the case.

You can therefore leave it as it is.

foldynl, 2022-06-29:
I leave it as it is now. But the theme about about has to be solved in the future because #81

closed 2022-06-29 by foldynl