foldynl/QLog #86: Online Map - Uncaught TypeError when network is unreachable

foldynl, 2022-06-17:
10:17:34.598 [CRITICAL][js]	=> Uncaught TypeError: L.terminator is not a function [(null):qrc:/res/map/onlinemap.html:48] 
10:17:34.681 [CRITICAL][js]	=> Uncaught TypeError: Cannot read property 'createIcon' of undefined [(null):qrc:/res/map/onlinemap.html:5] 
10:17:41.383 [CRITICAL][js]	=> Uncaught TypeError: L.terminator is not a function [(null):qrc:/res/map/onlinemap.html:48] 
10:17:41.441 [CRITICAL][js]	=> Uncaught TypeError: Cannot read property 'createIcon' of undefined [(null):qrc:/res/map/onlinemap.html:5]

pulponair, 2022-06-23:
@foldynl i can help

pulponair, 2022-06-23:
@foldynl 
Here we go.
There is more refactoring nessesary, you are using global variables to interact with
https://foldynl.github.io/QLog/js/L.MaidenheadConfWorked.js'

Have a look and tell me what you think. I am more than happy to help more with the map

closed 2024-06-08 by foldynl