foldynl/QLog #90: New QSO - 'Prefix - Callsign - Suffix'

DL2KI, 2022-06-24:
When entering a callsign in the 'New QSO' area, the QSO table will show the previous QSO's and the existing data will be transferred to the 'New QSO' area.

Example: DD5PX

If 'DD5PX' is portable QRV in Denmark, the callsign is 'OZ/DD5PX'.

If you now enter 'OZ/DD5PX' in the 'New QSO' area, the previous QSO's with 'DD5PX' will not be displayed in the QSO table and also the existing data will not be transferred to the 'New QSO' area.

With the input form 'Prefix - Callsign - Suffix' (example: 'OZ/DD5PX/mm') therefore only the callsign should be used for the display in the QSO table and the data transfer into the 'New QSO' area.

Location specific data can then be edited there. Fixed data will be preserved.

foldynl, 2022-06-24:
it's a good idea, but I'm not sure what should be taken from the previous QSO. 
It seems to me that only Name, QSL Via, email, not 100% sure URL (can change based on location), maybe My Notes, but also Notes can changed based on location. 

But what you are right that Prefix/Suffix should be omitted when you want do display a station Info.

DL2KI, 2022-06-24:
First you can see if you already had QSO's with the callsign. And QSL information or information under 'My Notes' are available.

The callsign is the primary identifier. If left as it is, it is a previously unknown QSO partner that is not associated with the regular callsign. 

There are also some OM's who have their own entries in QRZ.com. See for example 'OZ/DK1DU', 'OZ/DL9LBR' or 'DL8NBM/MM'.

From my point of view this information is not unimportant. Editing the QTH parameters, add information in 'My Notes' or adding e.g. IOTA data is not a big problem.

commit 7d0c938c (2022-07-15): Fixed #90 - Search Callsign - partial match

The main search widget was changed to search a callsing
with parter %callsing%.
commit 8b0e5554 (2022-07-15): Fixed #90 - QSO fields from prev QSOs when Prefix - Callsign - Suffix

QSO fields are filled based on information from prev QSOs. If QLog
looks for a match then it begins with a full callsign. If the full
callsign is not found then it tries to find a based callsign.

Full set of fields are taken only in case when QLog finds an exact
match for full callsign. Otherwise, it takes only Name, Notes,
URL and email.
commit ffcab7e9 (2022-07-15): Callsign class: Introduced callsign class and used by Fix #90
closed 2022-07-15 by foldynl