foldynl/QLog #97: Grayline-Map

DL2KI, 2022-07-05:
The new grayline map is now more recognizable. However, the grayline itself can only be guessed at with this day-night representation.

[image]

Since this line can be interesting for Grayline QSO's, such a Grayline map is much clearer from my point of view.

[image]
(https://dx.qsl.net/propagation/greyline.html)

But before you have long discussions about this with different users, you could perhaps implement an exchangeable map for later QLog versions. Then every user can use the graphic according to his own taste.

From my point of view it would be sufficient to store an alternatie map as a graphic file with a fixed name in the configuration folder. Then the user would have to copy his preferred graphic with this name into it.

QLog would then use the default graphic if no alternative graphic is available. This would also not require any additional settings.

foldynl, 2022-07-11:
I am just testing but...is it better? It is a modified map color level for the upcoming release.

[image]

DL2KI, 2022-07-12:
> but...is it better?

Hi,

from my point of view it is better because the grayline is much easier to see. Also, the land structures are more recognizable in the night zone. The representation of fictitious illuminations in the night zone has no advantage for the viewer in orientation.

But, as always, there may be other opinions on this. 

That is always the disadvantage of single opinions. That's why I suggested to copy an alternative graphic with a defined file name into the config folder, which would then be displayed as an alternative to the standard graphic.

But you can leave it all as it is for now, since it doesn't affect the function of QLog. If you want to change the graphic later, you can make it dependent on comments of other users.

dl2ki, 2023-04-18:
Can be closed from my point of view.

closed 2023-04-18 by foldynl