foldynl/QLog #100: "New QSO" window reset

DL2KI, 2022-07-13:
If a callsign is entered in the "New QSO" - window, e.g. by clicking in the bandmap, the callsign remains there when a band change (at the rig, or in the rig widget) is performed.

It would be worth considering whether and at which events (e.g. band change) the "New QSO" window is automatically reset. It is of course possible to reset the "New QSO" window manually by pressing "ESC".

It may also be that the current behavior gives an advantage. From my point of view rather not.

foldynl, 2022-07-13:
It is good note. I thought about it when I used QLog last weekend during IARU HF World Championship. But I don't know exactly under what conditions it should be done. 

N1MM approach is interesting, QSO parameters are deleted when the frequency changes. That seems logical to me. But I'm  not sure about the number of difference. It is possible to approach it in such a way that SSB or CW has a certain bandwidth. Unfortunately, during contest the stations are sometimes very close and it could cause false deletions. That's why I didn't implement it yet.

But this is definitely in the plan.

DL2KI, 2022-07-13:
Yes, exactly when to do it needs to be considered.

A callsign is entered in the NewQSO window, either by hand (when you hear the call on the set frequency), or by clicking in the bandmap. In both cases there is a link between callsign, band, frequency and mode.

Now it depends on whether the QSO was started or not.

If the QSO has not been started, from my point of view a reset can always occur if a change of the band takes place or if the frequency is changed by a larger amount. Fine tuning of the frequency, e.g. to ZeroBeat, must be possible. This reset can also be triggered manually by "ESC".

When the QSO has been started, it happens that a QSY occurs in the QSO. E.g. to avoid another QSO or interference. However, this QSY always takes place within narrow limits in the same band. (QSY on another band or another mode is always a new QSO).

Therefore a reset in the running QSO may not take place.

In QLog is missing from my point of view a clear, well visible indicator for the cases "QSO started" and "QSO not started. In CQRLog a separate QSO timer is shown for this.

In QLog the timers always run and it is not easy to recognize whether a QSO was started. Here it would be to be considered whether and how one changes this.

Examples:

QSO not started
[image]

QSO started
[image]

QSO not started
[image]

QSO started
[image]

or any another indicator (green dot, green triangle, ...)

Maybe you have to try out the criteria yourself first. You can also adjust the criteria again later.

foldynl, 2022-07-13:
Many thanks. You confirmed a few things for me. 

Currently, I have to finish upcoming 0.12 and then I will test some changes what you mentioned above.

DL2KI, 2022-07-14:
In this context, perhaps we should think again about introducing a clearly recognizable "Start-QSO", "Stop-QSO" button.

Shortcuts are OK, but they are only present in the head, not in the eye.

This is just an idea. I haven't looked here yet how this is solved in other logbook programs (except CQRLog).

I think you will come up with a good solution to the complete issue.

DL2KI, 2022-07-16:
> New QSO time start/stop description; added Automatic QSO Time start when RSTs is focused

The handling with this solution is good. I think that a "Start QSO", "Stop QSO" button is no longer necessary.

The green colored callsign from https://github.com/foldynl/QLog/issues/100#issuecomment-1183202483 would look visually overloaded. So also not such a good idea.

Green font in start timer after QSO start would be worth considering.

foldynl, 2022-07-18:
> The handling with this solution is good. I think that a "Start QSO", "Stop QSO" button is no longer necessary.

I know that they are not necessary but to be honest, I didn't have the courage to remove them. But I think that in the next release those buttons won't be there anymore.

> Green font in start timer after QSO start would be worth considering.

I agree but I have to test it. Maybe it won't be easy to read after the change, I'll have to test it.

foldynl, 2022-08-06:
QSY Wiping function and flashing Time Off were introduced in the new v0.13. I am closing the issue. Please open a new issue if something is missing.

closed 2022-08-06 by foldynl