foldynl/QLog #107: Entering paper log QSO's

DL2KI, 2022-09-06:
When QLog is running and connected to the station rig, QLog will lose this connection if you enter QSO's from a paper log and select another rig in the "My Station" tab that was used for the QSO.

To enter paper log QSO's you have to disconnect the connected rig before.

In addition, the QSO timer starts when you fill in the 'RSTs' field when entering a paiperlog QSO from the previous day. To enter the QSO time, the timer must then be stopped manually.

There is no problem to proceed this way, since one does not use the rig to enter paper logs at the same time. But all in all the procedure for entering paper log QSO's afterwards should be reconsidered. The number of OM's who do this, however, will probably be rather smaller.

foldynl, 2022-09-08:
> When QLog is running and connected to the station rig, QLog will lose this connection if you enter QSO's from a paper log and select another rig in the "My Station" tab that was used for the QSO. A change in QRG is transmitted to the connected rig.

What you described is an use-case for which QLog is not prepared. it's a use-case when you enter QSOs with a Rigs but you have another Rig connected (online). Can you please tell me under what circumstances this setup is used?

> In addition, the QSO timer starts when you fill in the 'RSTs' field when entering a paiperlog QSO from the previous day. To enter the QSO time, the timer must then be stopped manually.

You are right. It has to be, somehow, change. At this moment, I am not sure how to recognize whether Timer should be started because you started a QSO and when leave the timer stopped because it is a manually entered QSO. I don't like to introduced any "offline button" for it. Do you have any idea how to recognize the manually entered QSO?

DL2KI, 2022-09-08:
To enter manually logged QSO's there are two scenarios from my point of view.

1. one starts QLog only to enter some manually logged QSO's without connecting a rig.
2. one sits at the station, has the station rig connected and enters some manually logged QSO's besides.

In case "2" you have to disconnect the rig to be able to capture the manually logged QSO's, if for this purpose other rigs or another band / mode, ... were used.

In both cases, if manually logged QSO's were entered, no rig would be connected. This could perhaps be used as an "indicator" and the timer turned off. I am not sure about this yet though.

In the case of entering manually logged QSO's, the seconds in the timer are unnecessary in any case and could be set to "0".

Knowing this, you can manually enter logged QSO's, albeit a bit cumbersome.

After all, there are only a few QLog users at the moment, so this could be put on hold until there are more opinions on it. The manual input of QSO's should be generally possible in a log software. Even if this is probably only used by a small number of OM's. But I have currently no overview, how this is solved in other programs.

From my point of view an "Offline Mode" would be a possible solution to realize this, and to decouple the input from the rig. 

Maybe the new "Edit-Mode" for QSO's can be used for this. The window can then be specially designed for manual QSO entry if required. This would then require a menu or button "Enter manually logged QSO's".

foldynl, 2022-09-08:
Wolfgang, 

I'm trying to understand how you use it. I can't imagine a situation where I would have a Rig online and logging a QSO from another Rig. It's  possible that I missed something, but I don't know when and under what circumstances this could happen. If I'm not working with connected Rig and doing a QSO on another rig, it doesn't make sense to have the unused rig connected (online). Please try to describe it to me, why did you do it? Maybe we can find something to make it better.

DL2KI, 2022-09-08:
Example scenario:

In the last days I have put some QSO's into a paper log. These were made e.g. portable with two different other QRP rigs, for which there are entries in "Settings - Equipment - Rigs".

Now I am at the station, have connected my station rig and am calling CQ on 40m. But since nobody answers at the moment, I want to use the time and enter the QSO's from the paper log.

Now I enter a callsign from the paper log, jump to "RSTs" and the timer starts to run. Now I stop the timer and enter "RSTr". Then a frequency in the 60m band and the connected rig switches from 40m to 60m, although I want to continue listening on the set frequency in the 40m band. Switching back to 40m does not work, because then the one given callsign is deleted. So I stay on 60m.

Then I go to "My Station", select the used rig and QLog loses the connection to the station rig completely. Now I add the antenna used (from "Settings - Equipment - Antennas"), and the transmit power, and other information, like "QSL send", "My Notes", etc....

Then I can save the manually entered QSO, and reconnect the station rig.

Mostly I do then without the connection to the station rig and enter the remaining QSO's from the Paiperlog so. In parallel I can then listen to the original frequency in the 40m band with the station rig.

From my point of view the input of a QSO's with a CAT connection to the rig and the input of a manuall captured QSO's are two different things, which are independent from each other in the log program.

The input of manually captured QSO's must not have any influence on the connected rig.

However, my considerations and procedures have the great disadvantage that they are an individual opinion. Such things are highly dependent on the habits and individual applications of the users. 

It primarily concerns those users who are QRV with conventional modes (CW, SSB, FM) and with small QRP rigs portable (SOTA, IOTA, WWFF, ILLW, ...) and then transfer your QSO's manually into your log programme. Users of digital modes tend not to be affected here.

Therefore, the considerations and comments should not affect your work on CW functionality. That is certainly more important. You can therefore leave it as it is for the time being and wait to see what other opinions there are on the issue later.

commit efcc722a (2022-11-17): Fixed #107 - New Contact: Added Manual Mode
closed 2022-11-20 by foldynl