foldynl/QLog #110: DX-Cluster / Network reconnect

DL2KI, 2022-09-08:
I currently have a DSL problem here that needs to be fixed by the network provider. The DSL connection is occasionally interrupted and re-established.

When interrupted, QLog loses contact with the DX-Cluster server. The message appears:

[image]

After reconnecting the DSL connection, the DX-Cluster must be reconnected manually.

Perhaps it would make sense to automatically reconnect to the DX-Cluster after a network failure if an existing connection was interrupted. 

To do this, the state of the connection before the network failure would have to be saved and then a reconnect would be attempted in the background (without a message window) at a certain time interval. The same would apply to other network connections.

foldynl, 2022-09-08:
Thanks for comment. I solved it couple months ago but I was not able to find any reliable way how QT inform an application that internet connection is established. That's why I postpone it.

There is a difference between to have a connection to internet and to have a connection to network.

OK...I will try to find any reliable solution for it.

DL2KI, 2022-09-10:
In this context, there is also the problem that the connection to the callbooks are lost.
So far I could only fix this by restarting QLog.

An automatic reconnect of the internet connections would also be necessary here. Is it not possible to check this e.g. by a periodic "ping" to a certain internet address?

commit 1d74fb3b (2022-09-23): Fixed #110 - Callbook reconnect
closed 2022-09-29 by foldynl