foldynl/QLog #112: Wiki - Bandmap

DL2KI, 2022-09-11:
> If the operator double-clicks on the callsign then QLog "tunes" the frequency.

Here the text content would have to be supplemented on occasion.

Example:
```
A double-click on a callsign in the bandmap sets the rig to the corresponding frequency, enters the 
callsign in the input mask of the "New QSO" area in the main window and prepares QLog for the 
start of the QSO.

A single-click on a callsign also sets the rig to the corresponding frequency and activates the 
"Callsign Whisperer" function. The callsign appears above the callsign entry field.

The function of the "Callsign Whisperer" is described in more detail in the Wiki under "Main Window".

```

foldynl, 2022-09-23:
Wolfgang,

I am testing now what you describe above and unfortunately I do not see any difference between single-click and double click. Please, could you describe exactly what you see when single/double click occurs? I can see that "whisperer" is set always. Please, send me also your setup when it occurs (rig is online/offline, callsign field contains/does not contain any previous data etc.)

DL2KI, 2022-09-23:
Hi,

sorry, I had not tested that completely.

If the rig is not connected, there is no difference between single click and double click on a callsign in the bandmap. The callsign is added to the callsign field in the NewQSO area and also appears above in the "Whisperer area".

When the rig is connected, there is the described difference between single click and double click.

With the single click the callsign field is cleared in each case.

foldynl, 2022-09-24:
It is unwanted behavior. I am closing this issue. The bugfix will be solved as part of issue #116

closed 2022-09-24 by foldynl