foldynl/QLog #115: Callsign allert / Bandmap

DL2KI, 2022-09-15:
Clicking on a callsign displayed in the Alert area of the main window (bottom right) will switch the rig to the corresponding frequency. The data for the call is entered in the NewQSO area.

The bandmap switches to the corresponding band. However, the call will not be displayed in the bandmap if it is outside the current band range.

It would be worth considering centering the bandmap around the selected call for this case.

foldynl, 2022-09-19:
For me, it is important to find out why the callsign was out of band. Why did it happen like that? Are band's boundaries defined incorrectly? As I remember, you mentioned that QLog has an incorrect band boundaries for 5MHz band. Is this the case?

DL2KI, 2022-09-19:
No, that's all fine.

For example, if you zoom into the band map when the band occupancy is high, only part of the band is displayed in the band map. 
The user then selects the area of interest.

If a callsign is selected in the "Call alert" line that is outside this currently displayed area of the bandmap, you have to track the bandmap manually if you want to see the callsign there.

It is just an effect that I noticed. If you want to track the bandmap automatically in this case you have to decide. You can also leave it like this.

foldynl, 2022-10-06:
What do you think about the idea that frequency centering will be always on ? in the other word, the option "centering" frequency will be removed from the context menu and it will be always on?

DL2KI, 2022-10-06:
I thought about it.

It is often the case that you set a certain sub-area in the band map that you observe. E.g. areas where contests are preferred or not preferred. Or band sections, in which preferentially certain QSO partners are to be found.

By a forced centering of the Bndmap the settings would be lost.

> It would be worth considering centering the bandmap around the selected call for this case.

Deviating from my original thought to center the bandmap, I would now even suggest not to center the bandmap, but to shift the adjusted section of the bandmap in such a way that the clicked callsign is just visible at the top edge or at the bottom edge of the bandmap.

This way, the band section of the bandmap set by the user is affected the least.

-----

Personally, I prefer software that has a good default setting, but at the same time leaves many options open for customization.

This is one reason why I have been using the KDE desktop for my Linux systems for many years. The default settings in KDE are so well chosen that individual settings are rarely the case for me.

Therefore I am of the opinion that one should not reduce possible settings too much.

I know that there are also other opinions here. Some say that KDE has too many settings that overwhelm the user, others say that Gnome restricts the user too much.
 
In order not to overload the context menus, one could also think about combining the basic settings of the GUI in the general settings.

Basically, I belong to the group of users who prefer more settings than less. But I think that more user opinions are needed on these topics. Until then, you should continue on your current path.

commit f1adc126 (2022-10-10): Fixed #115 - Keeping the Bandmap RX mark always visible when centre RX is disabled
closed 2022-10-16 by foldynl