foldynl/QLog #116: Bandmap callsign select

DL2KI, 2022-09-19:
From my point of view there is a problem with the callsign selection in the bandap.

A single click on a callsign in the bandmap triggers the "Callsign Whisperer" function. Another single click on the same callsign transfers the callsign to the "New QSO" window.

A double click on a callsign in the bandmap, transfers the callsign directly into the "New QSO" window.

It happens that when you single-click on a callsign, the green cursor (triangles) does not jump to this callsign, but to another one.

In this case, the green cursor jumps to "9A/DL2HWI/P" and back a few times when i click on "EA4DOS/P".

[image]

In this case, the same between "G4AJY" and "J6MAT".

[image]

And in this case, the same between "W2MV" and "SP2FAP".

[image]

I have not yet succeeded in finding a connection or a systematic approach to the effects, since the problem does not always occur. Maybe try it out for yourself.

From my point of view, the "Callsign Whisperer" function in the bandmap could be completely omitted and only triggered when the callsign is dialed from the rig. This is the original idea behind this function. Possibly the selection problem is caused here.

The selection of a callsign in the bandmap should then be triggered either by a double click or by a single click, depending on what you prefer.

foldynl, 2022-09-23:
It is strange. I saw also your video what you sent me via email. I have tested and getting different behavior. During my testing, there was no difference between single and double click. see attached examples below when I used always a single click.
[image]
[image]

DL2KI, 2022-09-23:
Yes, strange.

Some things are hard to understand, especially when the behavior is not always reproducible.

I will continue to observe this. Maybe there will be something that can be used to find a better explanation of the effect.

foldynl, 2022-09-24:
Many thanks for your description in #112 and here. I found out how to test it. Unfortunately, the issue is connected with "whisperer".

commit 448f8a47 (2022-09-24): Fixed #116 - Bandmap callsign selection not fully works - single/double click difference
commit 797c2cef (2022-09-27): Fixed #116 - Bandmap callsign selection not fully works - single/double click difference (2)
closed 2022-09-29 by foldynl