foldynl/QLog #117: New QSO callbook data

DL2KI, 2022-09-20:
I have the impression that the display of the callbook data for a callsign is often not displayed or delayed.

When double-clicking on a callsign in the bandmap, the call is taken over in New QSO, the callbook data are not completed even after a longer waiting time. Entering the call directly into the NEW QSO window does not change anything here.

In other cases the callbook data appears relatively fast.

The behavior is unfortunately poorly reproducible and can of course also be due to the response behavior of the callbook servers. My settings are hamQTH (Primary) and QRZ.com (Secondary).

DL2KI, 2022-09-23:
The problem with my DSL connection has been solved in the meantime. So the internet is available all the time.

I cannot check the response time of the Callbook data query here. I will continue to monitor the behavior here.

It is possible that the effect is related to whether the call was found in the callbook or not.

Maybe an indicator "Call not found" could be inserted at a suitable place or 'not found' could be displayed in the field "Name" or a hint could be displayed in another form.

foldynl, 2022-09-24:
I am closing this issue. Please, open it again if you see the same behavior in the upcoming release 0.14 (it will available in the near future ;-) )

closed 2022-09-24 by foldynl