foldynl/QLog #120: CW settings, default "Key Mode

DL2KI, 2022-09-30:
For the CW settings, the "Key Mode" "IAMBIC B" is used as default. 

From my point of view the Iambic Mode "A" is more common here in Europe. 
Therefore it is to be considered whether one changes the default, or leaves so. Both variants are OK, because you can change it individually.

foldynl, 2022-10-01:
I thought about it and in the end I found only one article that deals with Iambic mode and brings some numbers about the usage of Iambic modes. The article is: [All About Squeeye-Keying (part 2), FOCUS - the Magazine of The First Class CW Operatots' Club, No. 127 Autumn 2021, str. 5](https://telegrafie.cz/files/sq2.pdf) where you can find a result of Iambic mode usage.

[image]

I know the 68 is not representative group for any statistic. Unfortunately I did not find better research. Please, is it possible to substantiate your statement with some statistics?

DL2KI, 2022-10-01:
Unfortunately, I don't have any statistical data on this either.

Telegraphists have been arguing for years about which method is the better one. It may depend on the country and which keyers were first established on the market there.

Before you risk philosophical discussions among users here, you should keep it that way, hi. It's no problem to set it individually.

But when you start with Sqeeze Keying, you have to choose a mode. Switching to another mode later is difficult.

I think you should close the issu until maybe more requests come later, or not ...

closed 2022-10-01 by foldynl