foldynl/QLog #124: CW Konsole, "Halt" Button

DL2KI, 2022-09-30:
The Halt button does not work for "Morse Over CAT". Is it possible to change this?
But it is no problem to cancel the transmission with a short press on a key padddle.

foldynl, 2022-10-01:
The Halt button is little bit magic. This feature is supported by Hamlib version 4.x and newer - it is not implemented in Hamlib 3.x. Therefore the button is disabled when Hamlib 3.x is installed. 

Unfortunately, I seems that Hamlib version autodetection under QLog does not work properly in case of .deb packages (rpm and windows packages should not be affected).

I will fix it  in the the version.

closed 2022-10-02 by foldynl

foldynl, 2022-10-02:
Fixed in a new release 0.14.1 which was only released for Ubuntu. No other platforms were affected

commit 6587ede5 (2022-10-02): Fixed #124 - CW Console HALT Button is not enabled under Ubuntu flavours