foldynl/QLog #125: CW console window

DL2KI, 2022-09-30:
The main window of the CW console takes up a relatively large amount of space due to its size, depending on the positioning of the windows. 

I have arranged the CW console as a switchable TAB, together with "Map" and "Online Map". This way I can use the windows alternatively without taking up additional space.

[image]

Maybe this possibility should be mentioned again especially in the wiki, because the individual window arrangement is a special feature of QLog.

If only the shortcuts are used, the subareas in which the transmitted CW text is displayed and the subarea in which the echoed CW text is displayed are not absolutely necessary, since the CW text is heard by the ear.

Therefore one can consider to add an alternative window "CW-Shortcuts", where only the shortcuts, the two combo boxes and the CW speed are displayed.

But the arrangement of the CW console as TAB is also a way to reduce the space requirement.

foldynl, 2022-10-01:
the design of the window had several versions - that's why it took so long to release a new QLog version.I think this one was the best of all the proposals. I don't want to separate the console and shortcuts, and changing the font size - I don't know if this will save enough space.

Therefore I leave it as is and we will see in the future.

dl2ki, 2023-04-18:
Can be closed from my point of view.

closed 2023-04-18 by foldynl