foldynl/QLog #133: Callbook behavior

DL2KI, 2022-11-24:
In the Settings / Callbook I have set HamQTH as primary and QRZ.com as secondary.

Now I have noticed the following behavior:

When clicking on the call "CT9/DL5CW" in the bandmap, the callbook data (Name) for the call is not shown. Even if  I enter the call by hand, this is the case.

When entering "DL5CW" the callbook data is shown.

https://user-images.githubusercontent.com/26311402/203842714-e88333f2-537c-4e72-af4a-4507370b8db1.mp4

If I enter e.g. "CT9/DM2DZM" the callbook data (Name) is shown correctly.

https://user-images.githubusercontent.com/26311402/203842724-6f702a2d-52d0-448b-8ce3-24ebc0e17b56.mp4

If I enter the call "CT9/DL5CW" in HamQTH, the data is displayed correctly.

The question arises why the name from the callbook is not displayed for the call "CT9/DL5CW" and why the name is displayed for the call "CT9/DM2DZM".

This could also be a mistake in my thinking.

foldynl, 2022-11-25:
I can confirm the similar "strange" behavior when I try it with my QLog

what's worse, even CT9/DM2DZM doesn't work for me with different callbook combinations

DL2KI, 2022-11-25:
Sorry, my mistake!

I just watched the video 2 again and found the answer there. 
"CT9/DM2DZM" works here, because the call is already in the log.

It seems to be a smaller problem with the callsign query in the callbook.

foldynl, 2022-11-25:
it seems that this bug has been there since the beginning of the implementation. It is about searching for calls with a prefixes(suffixes). 

Unfortunately, just now I find out that QRZ and HamQTH automatically remove the prefix(suffix) and return the base call. It means, you query CT9/DM2DZM, but the result is automatically returned for DM2DZM (if exists). And that's the issues why QLog doesn't show anything, because QLog checks if the requested callsign = returned callsign. 

I will have to implement a partial match when receiving the search result.

commit 9e224468 (2022-12-18): Fixed #133 - Missing callbook data when callsign has prefix
closed 2022-12-18 by foldynl