foldynl/QLog #138: User defined rig control buttons

DL2KI, 2022-12-02:
Another idea that could be taken from CQRLog is a small number of "user defined rig control buttons". In my opinion, the number of 3 buttons is sufficient to not overload the rig window.

With these you can e.g. trigger the auto-tuner on the rig after a band change, or other functions such as CW spot, which is often needed with narrow CW filter bandwidths.

Example:

[image]

foldynl, 2022-12-04:
my primary thought was that the Rig widget was not a full-fledged control panel for the Rig. Rig widget should be a secondary Rig display. To control the rig, an application specialized for Rig control or the rig itself should be used. For example [wfview](https://wfview.org/) for ICOM rigs. I have a feeling there is also something for Yaesu. I don't know if there is something for Kx rigs as well.

I can never get close to this specialized Software in terms of control in QLog at the price of starting to increase a size of the Rig widget at the expense of other widgets.

DL2KI, 2022-12-04:
No problem. It's good if you take a consistent approach to the design of the GUI.

When I introduce some ideas for additional functionalities here, it is mostly based on things that just come up for me in practical radio operation. 

If you click on a callsign in the band map and transfer it to the QSO field, retuning the rigs to "zero beat" is almost always necessary with narrow filter bandwidths. However, this can be done just as well directly on the rig, which is directly within range.

foldynl, 2022-12-04:
I really appreciate your feedback. I had the intention of going this way to implement a rig control via QLog (so that I don't have to "touch" my rig on the table, hi). But I considered what kind of software QLog actually is. 

I tried several specialized control software and thought that QLog should be focused on what it was intended for - QSO logging.

Specialized software for rig control are able to fully adapt to a given rig. If I implement the full rig control in QLog, it would always be a compromise, because the control would have to be very general (there are many rigs with different interface and different hamlib properties, for some rigs hamlib implements very very elementary things, for some of them can fully control all features).

I'll leave the issue opened if we have any other thoughts on this topic. But I don't plan to implement it at this time.

closed 2023-04-18 by foldynl