foldynl/QLog #139: CW speed synchronisation

DL2KI, 2022-12-02:
When using the WinKey interface, the CW speed can be set on the Winkeyer (e.g. rotary control) or in the CW console. This setting does not affect the CW speed setting on the rig.

The rig is therefore keyed with the CW console via the WinKeyer and otherwise via the key connected to the rig.

If you use the CW console, e.g. for CQ calls, but then want to continue the QSO itself by hand at the same speed, the CW speed on the rig may have to be adjusted to that of the WinKeyer, as the speed set on the rig is usually not directly visible.

One possibility would be an automatic synchronisation between both settings, if this is possible for the rig via CAT. You should be able to select this synchronisation in the Settings / Rig, if desired, when a WinKeyer is selected there.

This would optionally ensure a uniform setting of the CW speed when using a WinKeyer.

This is merely an idea that has arisen today from practical QSO operation. You would have to consider whether such a synchronisation could be useful.

foldynl, 2022-12-04:
As always, I have a few questions. After the first reading, I said to myself that it is clear, that it must be done. But as I think about it, I would like to ask this:

If I understood correctly, your key is connected to the rig, the Winkey is sitting between the QLog and your rig. Why don't you connect the key directly to Winkey? This means the key is connected to the Winkey and the Winkey is between the QLog and the rig. In that case, there is no need to synchronize anything, because speed is send only to Winkey. Then you will have the same speed for QLog and key.

I have the second question. If the Rig can set CW speed via CAT, can't it also send CW via CAT? In that case, there is no need to have Winkey.

Thanks for the answer.

DL2KI, 2022-12-04:
yes, you can also connect the paddle directly to the Winkeyer. This eliminates the "problem".

However, many users are used to certain keyers. These can be external keyers or internal keyers of the rigs. The keyers all differ minimally in timing and feel. In my case, the error rate increases at high CW speeds when I use the WinKeyer. That's why I prefer the internal keyer with the K3 and the K2 (but that's individual). 

CW of course also works via the CAT interface of the K3 and K2. Therefore, the use of my WinKeyer is not absolutely necessary. Otherwise it would be placed in a cabinet here, hi. But there are many, especially under Windows, who use or have to use a Winkeyer.

There is forum post on the subject in the CQRLog forum, but it has not been answered. https://cqrlog.com/node/435

In principle, optional synchronisation between WinKeyer and Rig is of course not necessary. It also works without it.

This is therefore another question where you would have to hear several other opinions. A short-term implementation is not necessary. Perhaps move it to the "queue".

DL2KI, 2022-12-04:
The KX2 and KX3 can also be used without the accessory paddle. I think it is a very individual question whether such a feature is used or not.

commit 6e43ced7 (2022-12-13): Fixed #139 - CW speed synchronisation

CW Speed is synchronized in case when connected key
is not MorseOverCAT and operator enabled this feature in Rig Setting

this synchronization can be used if, for example, Winkey is connected
to the rig and the rig (eg K3X) has a build-in key.
Then an operator can begin QSO via winkey or QLog and finish it via  build-in key
in the same CW Key speed.
closed 2022-12-18 by foldynl