foldynl/QLog #147: Show Alerts Window

DL2KI, 2022-12-30:
Which criterias are used to display the recorded allerts in the window? 

Example:

This alert is displayed in the footer of the main window. Time approx. 16:50 UTC.

[image]

When the "Show Alerts" window is opened, the entry appears at the second position in the list.

[image]

Three alert rules are active.

foldynl, 2022-12-31:
The logic is that if the callsign does not exist, it is added to the first position. If it already exists, only the "Updated" count is updated (its position is not changed)

DL2KI, 2022-12-31:
Yes, thank you for the information. 

You can do it that way. No idea what logics there are and whether one is better than the other, hi. It is important to know how it works.

foldynl, 2023-01-01:
the logic is that I tried several approaches to display the data. I also tried the approach that dates are displayed according to time. Unfortunately, all variants led to the fact that they were unclear.

It is necessary to know that data can change very quickly - especially in the case of RBN. In these cases, it is necessary to keep order -the  reordering the records after each update  leads to chaos. Therefore, I chose the approach.

DL2KI, 2023-01-01:
Yes, the display of the records is OK like this. It was only about understanding how the arrangement comes about.
Maybe you can give a hint in the wiki.

foldynl, 2023-01-01:
will be updated in the WIKI

closed 2023-01-15 by foldynl