foldynl/QLog #150: New QSO, Locator field

DL2KI, 2022-12-31:
I had a QSO with an OM who was QRV portable at the "GMA DA/BW-282" location.

https://www.cqgma.org/zinfo.php?ref=DA/BW-282

I copied and pasted the locator "JN48ES91CA" into the New-QSO window. 

In the QSO list the 10 character locator is displayed correctly. In the QSO detail window also.

[image]

Likewise in the database.

[image]

In the QSO detail window the locator is displayed correctly also.

[image]

In Edit mode a warning appears: "Warning: DX Grid has an incorrect format".

[image]

foldynl, 2023-01-01:
the issue is more complex.

QLogs, including the last version 0.17, support only 6-chars locator.  it's my mistake because I misread the ADIF specification. Currently QLog supports ADIF v3.1.3 and ADIF v.3.1.3 supports the 8-chars locator. Therefore, QLog contains an known issue because it does not support the 8-chars locator.

In your case of 10-chars locator, the situation is different. ADIF v3.1.3 does not support such a long locator. The change occurred with the arrival of ADIF version 3.1.4 (released 6.12.2022), where the 10 and 12-chars locator support was introduced. 

I regularly adopt changes from ADIF specification after the release of the specification. This means that ADIF v3.1.4 will be implemented in the next version of QLog (I have currently prepared the changes in the Git repo)

Unfortunately, I decided:

- QLog will be fixed to support  8-chars locator (currently only 6-char locator is support)
- QLog will allow to import 10 and 12-chars locators but internally it will support only 8-chars locator for QSOs

Here it is necessary to explain why I made this decision to support only 8-chars even though the new ADIF support 12-chars locators.
 
- The approach of the author of the ADIF standard to 10-12 locators is very unusual. I am not at all able to understand the authors of this standard, why the support of 10-12 locators is defined as it is defined in 3.1.4. Just to imagine. The new ADIF introduced a special field (GRIDSQUARE_EXT) for the locator, where only characters 9, 10, 11, 12 are stored. Characters 1-8 are stored in the already defined field (GRIDSQUARE). In my opinion, it is a crazy decision and I hope that it will be changed or clarified in the following ADIF versions.
- until now, I thought that support for an 8-chars locator is sufficient, if you are not working on the UHF band (the precision is approximately +- 900m for 8-chars locator). I have one example. 10-chars locator defines a location with precision 1.25 seconds longitude and 0.625 seconds latitude (1 second longitude = 30m), 12-chars locator defines a location with precision 0.125 seconds longitude and 0.0625 second latitude (approximately 3.75m). that's an accuracy that I think I cannot be realistically measured, and it's also an accuracy that can't currently be displayed by QLog.

I know I'm very critical, but the authors of the ADIF standard should be careful. To be honest, the ADIF standard should basically be revised. For example, the approach of defining a field for every online service is an approach that is very poor. There are also many fields that contain redundant information or information that can be derived from the different field. This approach leads to the fact that the ADIF standard contains around 160 fields, and it is necessary to be very careful that the fields are consistent with each other - therefore QLog must contain many cross-validations when you edit the QSO record.

I could talk more about ADIF and my opinions , but it would be a long conversation.

In conclusion, I would sum it up like this:
- upcoming QLog will support ADIF 3.1.4.
- QLog will allow to import 10, 12 grids but QSO will allow to insert only 8-chars grids
- if there is a larger number of users who want to introduce 10, 12-chars locators for QSOs, it will be implemented with corresponding changes in distance calculation and distance display.

foldynl, 2023-01-01:
I have forgotten. Please change your locator to 8-chars. QLog would generate incorrect ADIF output in your case.

foldynl, 2023-01-01:
Just adding a link, where 12-chars grid is discussed. Really should have used the lat/lon information that is already in the ADIF standard instead of using a new field for  very precisely determined locations

https://groups.io/g/adifdev/topic/gridsquare_my_gridsquare/87937006?p=Created%2C%2C%2C20%2C1%2C0%2C0&jump=1

DL2KI, 2023-01-01:
Thank you for the detailed explanation.

I myself do not think 10-digit locator is necessary. It may be that there are use cases for this, when one is looking for a certain point in a map. For a logbook program, 6 or 8 digits are quite sufficient in my view.

However, the problem is not the locator itself, but the fact that you can enter it in the locator field and that it is also transferred to the database.

It would certainly be more consistent to shorten the locator to the maximum allowed length directly after entering it.

foldynl, 2023-01-01:
> However, the problem is not the locator itself, but the fact that you can enter it in the locator field and that it is also transferred to the database.

Yes. It is an issue that will be fixed

foldynl, 2023-01-16:
Closing the issue. QLog support locator up to 8-chars. At this moment, 10 and 12-chars locator is not planned. If there are more requests, the issue will be reopened.

closed 2023-01-16 by foldynl