foldynl/QLog #159: Bug report: Hamlib connection does not enable UDP port

rsaue, 2023-01-13:
hi @foldynl,

When using Qlog with Hamlib seems to disable the default 4532 UDP port in Hamlib, thus making it impossible for other programs to access the Hamlib connection with UDP using the Hamlib NET rigctl option. 

Connecting Qlog to my FTdx10 using HAmlib, should normally allow JTDX/WSJT to connect using UDP port 3432.  

I have tested in Linux and on Windows with the same result. 

Testing using CQRLOG on Linux and LogHX on win11 as main program on Hamlib to the Rig, enables connection for JTDX using "Hamlib Net rigctl". 

Time beeing I am using FLRIG to connect directly to the RIG and connecting QLog and JTDX to the FLRIG server using Hamlib FLRIG option. 

I feel it should be possible to do all this without using FLRIG. 

73 de LB2EG Richard

foldynl, 2023-01-13:
I am attaching an email conversation between Wolfgang (DL2KI) and me. This conversation may clarify the situation in QLog.
```
Hi Ladislav,

a question not directly related to QLog to a professional.

I run my K2 via a USB-Serial converter on the interface "/dev/ttyUSB2".

Now I'm looking at a way to use a SDR software in parallel to QLog to
use a SDR receiver as a panadaper. The CAT information should be
available to both programs. If I access the same interface with another
software while QLog is running, there is a conflict between both programs.

My question is, is there a possibility to access the serial device with
an additional program under Linux?
Is it possible that QLog itself provides such an additional virtual
interface?

On Windows you can apparently use a "Virtual Com Port" software for
this. Under Linux, however, I have not yet found a way to do this.

I think this can be interesting for other users as well.

73, Wolfgang
DL2KI

```

```
Hi Wolfgang,

Currently QLog does not support to share the Serial Port.

I did some research, but I didn't go into detail. I'll try to wonder about it.

Regards
Ladislav

I'm sorry, I sent the email earlier than I wanted.

I wanted to add that I also solved this, but in the end I do it by using rigctld and pointing both SWs (if it can do it) to this "rigctl server". It is the simplest and fastest solution. Everything else is not very "multiplatform".

Solving this in QLog is not very solvable. QLog runs on Win, Mac, Linux. Accessing and sharing the Serial port is handled differently on each platform. That's why I chose a simple path for myself - via rigcltd.

Regards
Ladislav
```

```
Hi Ladislav,

Thank you very much.

The variant about "rigcltd" I had also tried once at CQRLog. I will look at that again more closely and try for my application.

73, Wolfgang
DL2KI
```
```
Wolfgang,

Configuration is quite simple.

Run rigctld with parameters for your Rig. in my case, I am using this command:

rigctld -m 3073 -r  /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_IC-7300_03018006-if00-port0 -s 115200 --set-conf=data_bits=8,stop_bits=2,serial_parity=None,serial_handshake=None  -vv

-m 3073 .... ICOM7300 ID
-r .... the path to Com Port (I am using "/dev/serial/by-id/" Com Port name to prevent changes when disconnecting and connecting the port
-s ... Com Port baudrate
--set-conf - configuration parameters for Com Port
--vv ...verbose level (just for my information) it is not necessary to use it.

And in the applications you set the type of Rig "HamLib NET Rigctl" and the IP address (usually localhost but sometime I use also remote QTH therefore I use VPN IP address)
```

```
Hi Ladislav,

thanks for the additional information. I have done this according to the instructions

https://www.cqrlog.com/help/img/setting_rigctld_for_all_programs.pdf

and it works fine with QLog (with one rig).

I found an older post on the CQRLog forum asking about using 2 different rigs.

https://www.cqrlog.com/node/2513

The procedure with different ports does not work with QLog, because port 4532 cannot be changed in the settings.

> QLog-Wiki:
>
> Network attached devices are defined by:
>
> Host name - a host name of the devices.
> Port - number of port. Currently, Hamlib does not support to change a port number therefore it is not possible to change the default 4532.

In post #2 OH1KH writes that he uses rigctld with 2 rigs and ports 4532 and 14532. Could you occasionally check whether rigctld in QLog also works with other ports? But there is no hurry.

The example script would then have to be supplemented to check whether and which rig(s) are found at the specified interface. Otherwise, the start of rigctld terminates with an error message. The general start of the example script via crontab does not seem to be appropriate in this way.

OK, Ladislav. I'll see if I can think of a more practical solution.

73, Wolfgang
DL2KI
```

```
Wolfgang,

QLog does not execute rigctld as CQRlog does. QLog uses HAMLIB to directly call Rig's function.

If you want to run rigctld then you have to run it externally. It means that you have to run rigctld and then run QLog or other applications. 

Regards
Ladislav
```

rsaue, 2023-01-13:
Dear Ladislav,

Ok, now I understand, .QLog does not invoke the hamlib rigctld program.  

I can run this program as a script as you described. 

This is quite easy in Linux.  But not straight forward in windows. I guess I will stick with the FLRIG solution. 

Many hams run FT8 using WSJTX/JTDX. When they realize that they cannot do this out of the box with Qlog, I am afraid there will be some disappointment. 

73 de LB2EG Richard

foldynl, 2023-01-13:
Unfortunately, I don't have a simple and multiplatform solution at this moment. The best solution would be to implement rigctld directly inside QLog.

foldynl, 2026-03-13:
It is part of 0.49.