foldynl/QLog #176: Import ADIF

magnum88, 2023-02-22:
Hi
when import adif
on 323 QSO only 1 imported
the first line file adi
the other 322 no imported in QLOG
no error present on the software qlog and no error on my file beacause it's the same file for QRZ.COM
Same problem on package installed or source compiled
QLog 0.19.0
Based on Qt 5.15.3
Hamlib 4.5~git
Ubuntu 22.04.2 LTS x86_64

foldynl, 2023-02-22:
Please, could you send me the file or just the first 10 QSOs from the file?

magnum88, 2023-02-22:
Hi
i send you the file 10 lignes
73
Xavier

ADIF v3.1.0 Export from KLog
https://www.klog.xyz/klog
<PROGRAMVERSION:5>2.3.1
<PROGRAMID:4>KLOG 
<APP_KLOG_QSOS:1>8
<APP_KLOG_LOG_DATE_EXPORT:13>20230222-1244
<EOH>
<CALL:8>SO1980LL <QSO_DATE:8>20210714 <TIME_ON:6>095800 <FREQ:6>28.485 <BAND:3>10M <FREQ_RX:6>28.485 <BAND_RX:3>10M <MODE:3>SSB <MY_GRIDSQUARE:8>JN33MS92 <STATION_CALLSIGN:5>F4IYT <CQZ:2>15 <ITUZ:2>28 <DXCC:3>269 <COMMENT:36>MCHF 5W PAC12 COL DE VENCE (06) 963M <CONT:2>EU <EQSL_QSLSDATE:8>20210717 <EQSL_QSL_SENT:1>Y <LOTW_QSLSDATE:8>20210717 <LOTW_QSL_SENT:1>Q <CLUBLOG_QSO_UPLOAD_DATE:8>20210717 <CLUBLOG_QSO_UPLOAD_STATUS:1>M <NAME:8>LUBELSKI <QTH:6>LUBLIN <RST_SENT:2>59 <RST_RCVD:2>59 <TX_PWR:1>5 <APP_KLOG_LOGN:1>1 <EOR>
<CALL:5>DM1YS <QSO_DATE:8>20210715 <TIME_ON:6>080000 <FREQ:6>14.074 <BAND:3>20M <FREQ_RX:6>14.074 <BAND_RX:3>20M <MODE:3>FT8 <MY_GRIDSQUARE:8>JN33NO21 <STATION_CALLSIGN:5>F4IYT <CQZ:2>14 <ITUZ:2>28 <DXCC:3>230 <COMMENT:17>MCHF 5W EFHW 49:1 <CONT:2>EU <EQSL_QSLSDATE:8>20210717 <EQSL_QSL_SENT:1>Q <LOTW_QSLSDATE:8>20210717 <LOTW_QSL_SENT:1>Q <CLUBLOG_QSO_UPLOAD_DATE:8>20210717 <CLUBLOG_QSO_UPLOAD_STATUS:1>M <NAME:7>JUERGEN <QTH:21>88693 DEGGENHAUSERTAL <RST_SENT:1>0 <RST_RCVD:1>0 <TX_PWR:1>5 <APP_KLOG_LOGN:1>1 <EOR>
<CALL:5>F1PSX <QSO_DATE:8>20210716 <TIME_ON:6>093500 <FREQ:6>14.074 <BAND:3>20M <FREQ_RX:6>14.074 <BAND_RX:3>20M <MODE:3>FT8 <MY_GRIDSQUARE:8>JN33NO21 <STATION_CALLSIGN:5>F4IYT <CQZ:2>14 <ITUZ:2>27 <DXCC:3>227 <COMMENT:17>MCHF 5W EFHW 49:1 <CONT:2>EU <EQSL_QSLSDATE:8>20210717 <EQSL_QSL_SENT:1>Q <LOTW_QSLSDATE:8>20210717 <LOTW_QSL_SENT:1>Q <CLUBLOG_QSO_UPLOAD_DATE:8>20210717 <CLUBLOG_QSO_UPLOAD_STATUS:1>M <NAME:9>JEAN-YVES <QTH:7>LE MANS <RST_SENT:1>0 <RST_RCVD:1>0 <TX_PWR:1>5 <APP_KLOG_LOGN:1>1 <EOR>
<CALL:6>DL5ZBA <QSO_DATE:8>20210716 <TIME_ON:6>093700 <FREQ:6>14.074 <BAND:3>20M <FREQ_RX:6>14.074 <BAND_RX:3>20M <MODE:3>FT8 <MY_GRIDSQUARE:8>JN33NO21 <STATION_CALLSIGN:5>F4IYT <CQZ:2>14 <ITUZ:2>28 <DXCC:3>230 <COMMENT:17>MCHF 5W EFHW 49:1 <CONT:2>EU <EQSL_QSLSDATE:8>20210717 <EQSL_QSL_SENT:1>Q <LOTW_QSLSDATE:8>20210717 <LOTW_QSL_SENT:1>Q <CLUBLOG_QSO_UPLOAD_DATE:8>20210717 <CLUBLOG_QSO_UPLOAD_STATUS:1>M <NAME:3>UWE <QTH:12>35091 COELBE <RST_SENT:1>0 <RST_RCVD:1>0 <TX_PWR:1>5 <APP_KLOG_LOGN:1>1 <EOR>
<CALL:5>PD7RF <QSO_DATE:8>20210717 <TIME_ON:6>051300 <FREQ:6>14.074 <BAND:3>20M <FREQ_RX:6>14.074 <BAND_RX:3>20M <MODE:3>FT8 <MY_GRIDSQUARE:8>JN33NO21 <STATION_CALLSIGN:5>F4IYT <CQZ:2>14 <ITUZ:2>27 <DXCC:3>263 <COMMENT:17>MCHF 5W EFHW 49:1 <CONT:2>EU <EQSL_QSLSDATE:8>20210717 <EQSL_QSL_SENT:1>Q <LOTW_QSLSDATE:8>20210717 <LOTW_QSL_SENT:1>Q <CLUBLOG_QSO_UPLOAD_DATE:8>20210717 <CLUBLOG_QSO_UPLOAD_STATUS:1>M <NAME:4>ROEL <QTH:9>HILVERSUM <RST_SENT:1>0 <RST_RCVD:1>0 <TX_PWR:1>5 <APP_KLOG_LOGN:1>1 <EOR>
<CALL:5>DM2HK <QSO_DATE:8>20210717 <TIME_ON:6>062800 <FREQ:6>14.074 <BAND:3>20M <FREQ_RX:6>14.074 <BAND_RX:3>20M <MODE:3>FT8 <MY_GRIDSQUARE:8>JN33NO21 <STATION_CALLSIGN:5>F4IYT <CQZ:2>14 <ITUZ:2>28 <DXCC:3>230 <COMMENT:17>MCHF 5W EFHW 49:1 <CONT:2>EU <EQSL_QSLSDATE:8>20210717 <EQSL_QSL_SENT:1>Q <LOTW_QSLSDATE:8>20210717 <LOTW_QSL_SENT:1>Q <CLUBLOG_QSO_UPLOAD_DATE:8>20210717 <CLUBLOG_QSO_UPLOAD_STATUS:1>M <NAME:7>HARTMUT <QTH:7>ZEUTHEN <RST_SENT:1>0 <RST_RCVD:1>0 <TX_PWR:1>5 <APP_KLOG_LOGN:1>1 <EOR>
<CALL:5>F6HFT <QSO_DATE:8>20210717 <TIME_ON:6>071200 <FREQ:3>432 <BAND:4>70CM <FREQ_RX:3>432 <BAND_RX:4>70CM <MODE:12>DIGITALVOICE <MY_GRIDSQUARE:8>JN33NO21 <STATION_CALLSIGN:5>F4IYT <CQZ:2>14 <ITUZ:2>27 <DXCC:3>227 <CONT:2>EU <EQSL_QSLSDATE:8>20210717 <EQSL_QSL_SENT:1>Q <LOTW_QSLSDATE:8>20210717 <LOTW_QSL_SENT:1>Q <CLUBLOG_QSO_UPLOAD_DATE:8>20210717 <CLUBLOG_QSO_UPLOAD_STATUS:1>M <NAME:10>ANNE-MARIE <QTH:8>MULHOUSE <RST_SENT:2>59 <RST_RCVD:2>59 <TX_PWR:1>5 <APP_KLOG_LOGN:1>1 <EOR>
<CALL:5>LB8OH <QSO_DATE:8>20210720 <TIME_ON:6>215429 <FREQ:2>14 <BAND:3>20M <FREQ_RX:2>14 <BAND_RX:3>20M <MODE:3>FT8 <MY_GRIDSQUARE:8>JN33NO21 <STATION_CALLSIGN:5>F4IYT <CQZ:2>14 <ITUZ:2>18 <DXCC:3>266 <COMMENT:17>MCHF 5W EFHW 49:1 <CONT:2>EU <EQSL_QSLSDATE:8>20210720 <EQSL_QSL_SENT:1>Q <LOTW_QSLSDATE:8>20210720 <LOTW_QSL_SENT:1>Q <CLUBLOG_QSO_UPLOAD_DATE:8>20210720 <CLUBLOG_QSO_UPLOAD_STATUS:1>M <RST_SENT:1>0 <RST_RCVD:1>0 <TX_PWR:1>5 <APP_KLOG_LOGN:1>1 <EOR>

foldynl, 2023-02-22:
Many thanks for provided ADI records. They help me to identify many issues in QLog related to QSL handling.

Unfortunately, there are following issues in QLog:

1) QLog has incorrectly defined EQSL_QSL_SENT values because 'Q' value is missing in the list of supported values for EQSL_QSL_SENT. That is not critical issue because it can be change very quickly.
2) what is a more serious issue is that QLog interprets these values incorrectly. Unfortunately, the fix will be quite complicated. 

I'm sorry, but I don't have a quick fix for this now.

magnum88, 2023-02-22:
Thank's i make a change manuel on my file.adi

<EQSL_QSL_SENT:1>Q
by
<EQSL_QSL_SENT:1>N

i have my 323 QSO on QLOG
thank's
Xavier

foldynl, 2023-02-22:
Xavier,

the situation is not as simple as it may seem.

Yes, it is possible to give a workaround to change `Q` to something else. But during my analysis, I found out that I misunderstood the values of EQSL_QSL_SENT.

The behavior of the QLog is as follows now:
- QLog interprets `N` as "QSO not sent but should be uploaded", `Y` as "it was successfully uploaded". Other values are accepted, but have no special meaning. 
- QLog sends QSO to "online" service in case when qsl_sent is different from `Y` - what is totally wrong.

Based on ADIF spec:
`N` - do not send an outgoing QSL card; do not upload the QSO to the online service
`Y` - an outgoing QSL card has been sent; the QSO has been uploaded to, and accepted by, the online service
`Q` - an outgoing QSL card has been selected to be sent; a QSO has been selected to be uploaded to the online service
`R` - the contacted station has requested a QSL card; the contacted station has requested the QSO be uploaded to the online service

Qlog uses the default value `N` now. And here is the problem. If `N` means "not sent and will not send" , then QLog's issue is in the default value for QSL Sent value. I guess that the correct default value should be `Q`. And only `Q` and `R` QSOs should be uploaded to online service. If QSO is successfully uploaded to online log then the value should change to `Y`. `N` can be set by operator when QSO should never be uploaded to online service.

commit 13f49999 (2023-03-03): Fixed #176 - QSO Detail: Added posibility to change _SENT Status Values