foldynl/QLog #180: "Online Map", Language settings

dl2ki, 2023-02-28:
The online map displays country,- city names, etc. in the country's typical language (Arabic, Russian, Chinese, ...).
Is there a possibility to include layers here that show the labeling of the map in the language set in QLog?

[image]

[image]

[image]

dl2ki, 2023-02-28:
If this should work with the language layers, one could make the setting language of the OSM map also independently of the language setting of QLog. Then one would be flexible in any case.

foldynl, 2023-03-01:
Yes it's right. To be honest, I don't know how to fix it now. I have been googling about it for a while, but haven't come up with anything solid. It is possible to use German and French maps, but as far as I know, it is not possible to use the system locale. It is possible to solve it with some additional map layer, but I read that it has a significant impact on performance, because everything takes place on the client side. If you find out something interesting, please let me know.

dl2ki, 2023-03-01:
OK, thanks. I will do some research to see if there is a possibility here.

dl2ki, 2023-03-01:
I can not estimate what effort such an extension requires and whether it is necessary to implement that. 

Therefore, you can leave it as it is, because you do not always need the exact map details. You can see approximately where the QSO partner is located. In addition, the visualization of the OSM is, after all, the standard.

Independently of this, I have found an OSM map here, where the display language can be switched. For a more detailed look, if needed, you can also use such an external map, or Google-Maps, where the names are also displayed in Latin characters.

Example:
https://openmaptiles.org/languages/en/#4.68/37.96/102.76

https://openmaptiles.org/
https://github.com/openmaptiles/openmaptiles

I think that can be put on hold until you have nothing else to do, hi.

commit 79789a8a (2023-06-09): Fixed #180 - Online Map: Added alternative tile servers for DE and FR
closed 2023-06-09 by foldynl