foldynl/QLog #181: Jitsi Meet Videochat Screen Sharing

dl2ki, 2023-03-01:
When I started QLog and want to explain the settings and how it works to an OM in a Jitsi-Meet video chat, I can't enable sharing of the QLog window because the QLog window doesn't appear in the drop-down list.

When I open other programs, they appear in the window selection.

In the example, QLog and video chat are enabled. But only the Chromium window is offered for selection. Is it possible to change the QLog setting so that QLog is also displayed for selection?

[image]

System here is Kubuntu 22.04 LTS
- KDE Plasma version: 5.24.7
- KDE frameworks version: 5.98.0
- Qt-Version 5.15.3
- Kernel version: 5.15.0-60-generic (64-bit)
- Graphics platform: X11

foldynl, 2023-03-01:
"Ich habe keine Ahnung". I have no idea how the windows are selected. I'll try to look into it. But to be honest, I am not sure that this can be influenced from the application. Probably it is something between QT and window sharing implementation.

dl2ki, 2023-03-01:
I just tried this again with another QT application.

I have "QLog" running on desktop 1, "Firefox" on desktop 2, "QT Linguist v6" on desktop 3 and Chromium with Jitsi Chat on desktop 4.

In the window sharing selection window, the Chromium window, QT Linguist and Firefox tab appear. QLog is missing.
If the QT-Linguist window is displayed, shouldn't it work with the QLog window? They are both QT applications, after all.

You can easily try this out yourself in the browser. For me, no special window settings have been made in KDE/Plasma.

Let's see, maybe I can find a hint somewhere about the behavior.

foldynl, 2023-03-03:
Confirmed (not working) also under Ubuntu 22.04, Gnome 42.5, X11, Firefox and Chrome are impacted. 
It can be tested via https://webrtc.github.io/samples/src/content/getusermedia/getdisplaymedia/

foldynl, 2023-03-03:
workaround: use the full desktop sharing mode.

foldynl, 2023-03-08:
I did some tests but without any solution. I suspect one thing. Online QLog maps use QT Web browser, which uses Chromium. This Chromium should be hardware accelerated, which makes the problem with this sharing. I tried to implement a very simple QT application that uses the same settings and I didn't see it in the sharing window. I also tried various internal settings of the QT Chromium browser and it didn't help. I leave it opened now.

dl2ki, 2023-03-08:
No problem. This has nothing to do directly with QLog. I only noticed it because it did not work during a chat explaining the QLog settings. Eventually a solution will be found.

Thanks for your efforts.

mbridak, 2023-05-31:
Something may have changed... Works for me.

[image]

dl2ki, 2023-06-01:
On my system (see above) the QLog-Fenter still does not appear in the "Window" selection menu.
The selection menu for the windows also looks visually different here (see above).

foldynl, 2023-06-01:
No change in the QLog source code and no change in my case - I have no Qlog option in the Window Selection.
@mbridak : please, could you share your OS version (if it is Linux then share if you are using XOrg or Wayland) and Browser version?

foldynl, 2023-06-05:
It seems that Wayland is the answer. If I use chrome (the same version) + XOrg then I do not see the QLog in the window selection. If I use Chrome + Wayland then I can see the QLog in the window selection. 

So, a workaround can be using Wayland now.