foldynl/QLog #187: Rig widget, Band and Mode menu

dl2ki, 2023-03-10:
In the Rig widget, when the current seleted band or mode is in the lower list area in the "Band" and "Mode" combo box menus, the list expands upwards, beyond the frame of the main window.

[image]  [image]

No particular problem arises from this. You can also arrange different windows outside the main window. However, the question arises whether parts of the controls of any software should extend beyond the main window. Normally the main window should define the area in which a program is executed.

The selection menus in the filter area behave similarly.

This is just a general reference to a behavior of some menus that I noticed. I cannot judge whether changes are useful or necessary here. It does not affect the operation of QLog.

foldynl, 2023-03-10:
It is default behavior of ComboBox in QT. 

I don't see any problem with displaying the ComboBox content outside the main window if there is space. Personally, I don't know how exactly it should behave when the last item is selected and it is necessary to display this list in order, but I will try to see if there is another solution. 

It is possible to add Scrollbar on the left/right side but  I don't think it will bring anything clear.

dl2ki, 2023-03-10:
I would rather leave it as it is, since it does not affect the function. If there is not enough space at the top, the selection list moves to the bottom. With long lists (filter, country) the entries can be scrolled.

From my point of view, the issue can be closed.

closed 2023-03-14 by foldynl