foldynl/QLog #188: Online map behavior

dl2ki, 2023-03-11:
In some cases the online map shows only a partial path after clicking on a callsign in the bandmap. After a second click on the same callsign in the bandmap, the complete path is displayed.

**First klick**:
[image] 

**Second klick**:
[image]

**First klick**:
[image]   

**Second klick**:
 [image]

The display of the map switches from one display to the other when the callsign in the bandmap is clicked on again.

This behavior does not occur with all callsigns.

If additional information is required, please send us a short message.

foldynl, 2023-03-11:
I have some additional questions:

1) Do you use any secondary callbook ? (Settings->Callbook)
2) If you can see that map is switched then does the New Contact locator remain unchanged?
3) Are you able to find out what callsigns this is happening for? Are the callsigns something special?
4) If you enter manually the callsign in the New Contact field then do you see the same behavior?
5) DJ6ZM is a new callsign in your log or it is already present in your log?

I am not sure if we get something interesting but please, would it be possible to start QLog in debug mode for maps and send me the trace messages together with the callsign that caused this? But only for the time frame when the issue happens again.

Debug mode for maps:
`QT_LOGGING_RULES=qlog.ui.onlinemapwidget.*=true <path_to_qlog>/qlog`

foldynl, 2023-03-11:
I forgot one question. I guess you used Qlog v0.19. Please, confirm.

dl2ki, 2023-03-11:
> I guess you used Qlog v0.19

Yes, QLog is version 0.19

> Do you use any secondary callbook ? (Settings->Callbook)

    Primary callbook ist "HamQTH", secondary Callbook ist "QRZ.com"   

> If you can see that map is switched then does the New Contact locator remain unchanged?

     Yes, New Contact locator remain unchanged

> Are you able to find out what callsigns this is happening for? Are the callsigns something special?

      - Example Callsigns for which the map change occurs: DJ6ZM(*), IW2HTH, DK4AN(*), DJ3CQ(*), EA5GKA, DL9DBI(*)
      - Example Callsigns for which the map change not occurs: 9A3YT, MM0ZBH, LZ1SA
      
      Marked callsigns (*) are already in my log
      

> If you enter manually the callsign in the New Contact field then do you see the same behavior?

       When manually entering a call sign (which is currently also included in the bandmap) in the New Contact field, the map is displayed correctly. When clicking on the call in the bandmap, the map change occurs.

> DJ6ZM is a new callsign in your log or it is already present in your log?

       DJ6ZM is already in my log

The file with the debug data was created with the callsign DL9DBI where the map change occurs. DL9DBI is already in my log

[debug.zip](https://github.com/foldynl/QLog/files/10949082/debug.zip)

I can't see a relation of the effect with callsigns at the moment. To test it, it is best to use callsigns that are currently included in the bandmap.  I think the effect is in the code area of the bandmap.

foldynl, 2023-03-11:
Many thanks for the debug. The output looks pretty crazy. To clarify. What you sent me was created by clicking on a callsign in the bandmap. Is it correct?

Please, could you also send me your ADIF record for QSO with DL9DBI (you can send it to my email)? I try to simulate it, but I can't achieve this effect. Based on your debug output, it is definitely something bad with Bandmap or the New Contact.

foldynl, 2023-03-12:
Good news I am able to simulate it. Now just figure out what it is. Thanks for the export.

commit 3fa78a83 (2023-03-12): Fixed #188: Online Map incorrect Bounds if double-click on Bandmap callsign

QLog sends the position of the DXCC entity and then the exact position of
the station. In online maps, the leaflet.fitbounds function is used for each
of these changes in the Bounding box. The change is accompanied by a move animation.

Currently, it seems that if the position change comes earlier than the animation
is finished, the bounding box change is not performed properly.

Therefore the animation is turned off in QLog now.
commit a1936ab3 (2023-03-12): Fixed #188 - BandMap: Suppression of double tuneDX signal if double-click
closed 2023-03-14 by foldynl