foldynl/QLog #192: Some anomalies in the input and processing of " QSLr Date"

dl2ki, 2023-04-08:
Yesterday I got some new QSL cards, and entered the data in QLog.

For the first time I used the field "QSLRDATE" and entered the date in "Edit-Mode". In the field "Paper, Received" you can enter a date. The default in "Edit-Mode" is "02.01.00".

[image]

The default for "Edit Value" in the QSO list is "31.12.99". 

[image]

In "Edit-Mode" the input of the year is limited to 2 digits. Therefore I entered the date "07.04.23" in the field "Paper, Received". 

[image]

In the QLog database the date appears as "07.04.1923".

[image]

If I accept the default date "31.12.99" in the QSO list with "Edit Value", 

[image]

this value appears in the QLog database:

[image]

I can neither delete this date in the QSO list with "Edit Value", nor in the "Edit-Mode".

The pictures represent examples. Other manual entries in date fields have not yet been further checked.

foldynl, 2023-04-11:
in general, it is not recommended to use the date in the YYMMDD format after the year 2000,  for the reason you are sending now.  The year format is controlled by the environment settings. Therefore, it will be necessary to change your environment settings to the year format YYYYMMDD.

dl2ki, 2023-04-17:
My attempts to change the date format to 4-digit years were unsuccessful. I have not found a way to set this under Kubuntu 22.04.

According to my research, it is no longer possible to change the date format with KDE5. It is fixed in the "Locales".

I also haven't noticed yet that entering 4-digit year numbers is not possible in other programs. Programs like "KOrganizer" display the date correctly.

[image]

Therefore I cannot understand at the moment why QLog only allows 2-digit year numbers.

foldynl, 2023-04-18:
just for tracking the issue. EN locale has a shortDate year in format YYYY but DE locale has the shortDate year in format YY what caused the issue. QLog does not use the longDate format because it contains a lot of additional information. It seems so. that shortDate is very compact for QLog usage.

dl2ki, 2023-04-18:
Yes, but I have no idea about the internal structures in KDE, Plasma or QT applications.

Anyway, with 2-digit years, for example, you can't use "Date Range" in the statistics window either. 
I don't use these date fields at the moment. But in the database the dates of the QSO's are correct, so I can use QLog!

[image]

[image]

[image]

Perhaps a solution to the problem will be found later.

commit b1b6249c (2023-04-18): Fixed #192 - Added Time/Date Format for all fields
commit 3701939a (2023-05-07): Fixed #192 - Removed using YY date format - introduced own Locale Class
closed 2023-05-07 by foldynl