foldynl/QLog #194: CW Keyer Error

dl2ki, 2023-04-13:
When my WinKeyer is not connected, this error message appears.

[image]

I think that the text "No such file or directory" of the "Error Detail:" does not match the error. Presumably, the error message will refer to the keyer device.

foldynl, 2023-04-14:
Please, could you provide more detail when it occurs? Does it occur when Keyer is not connected and you have assigned the Keyer to the Rig which is connected? Or when does it exactly occur? The Detail error is correct because you define a file that represents the device and this file does not exist if the device is not connected. But it is important to know when this error occurs.

dl2ki, 2023-04-14:
The error occurred because the USB cable slipped out of the connector of the K3NG-Winkeyer yesterday, and I had not noticed it. If you now select "Equipment / Connect CW Keyer", the error message occurs.

The error message is okay from a technical point of view. Whether the user can do something with it is another question.

In this particular case, one could perhaps do without the detail line, as the problem with "Connectoon Error" is already described.  Or one could catch the error message and formulate the detail information for the pure user more understandably.

E.g. "CW-Keyer is not connected to QLog / PC", or similar.

commit 2e87cea2 (2023-04-25): Fixed #194 - Fixed user unfriendly CW Keyer Error
closed 2023-05-07 by foldynl