foldynl/QLog #201: Mode switch for ft857 is not active.

mppks, 2023-05-15:
The mode switch is not active in Rig Widget. 
[image]
[image]

mppks, 2023-05-15:
1. Yes, it's Windows 10, I used the EXE-installer
2. Yes, the switch is empty if I define a rig for the first time.
   2.1 If I restart Qlog nothing changes - the mode switch is empty.
3. If I delete all rig profiles, restart Qlog and create a new one, for example, for icom-7300, then the switch is still empty.
4. If the rig is disconnected (combo is disabled) - the switch is empty.

[image]

foldynl, 2023-05-16:
it is incomprehensible. I can't replicate this situation at all.

Can you please try another test. Run QLog with an input parameter ` -n test` (something like this: `qlog.exe -n test`)
This will cause qlog to start, but with a completely new database. Don't worry, you won't lose the original data, because if you run QLog again without a parameter, Qlog will return to the original database.

Once you have QLog running, fill in only the station profile in the initial settings and click OK. In this case, do you see "AM" in Rig Mode or not?

mppks, 2023-05-16:
Rig mode - empty.
Below is a link to the video from the screen.
https://drive.google.com/file/d/10yiz_6FbL1VkwlIgOK9jwgrVfSqTb6et/view?usp=sharing

foldynl, 2023-05-16:
Many thanks. The issue is confirmed. I can simulate it but I have to use Windows which is installed in Virtual Machine. That's very interesting.

mppks, 2023-05-16:
The Modes are working. tnx 73!

closed 2023-05-17 by foldynl