foldynl/QLog #203: LOTW, QRZ, ClubLog login issue

John-WE1DER, 2023-05-21:
I am having an issue with the above services not logging in properly. I enter in my user information for LOTW, ClubLog, and QRZ in the settings dialog and after clicking save and exiting the dialog, attempting to use LOTW is met with "LoTW Update failed: Incorrect Loging or password"; with ClubLog "Cannot upload the QSO(s): Error transferring https://clublog.org/putlogs.php - server replied: Forbidden"; with QRZ "QRZ.com is not configured properly. Please, use Settings dialog to configure it."
When I enter the settings dialog to check under the Sync & QSL tab, none of the passwords or QRZ api key are in the boxes.
Do I need to enter these passwords into a config file somewhere, or is this a bug that is not saving the login credentials?

foldynl, 2023-05-22:
Password Secure Store depends on the platform therefore please. could you specify your platform? Or you can send me a screenshot of the Main Menu->Help->About Dialog. 

Please, specify also, if you have installed it from package or from source code.

foldynl, 2023-05-22:
I assume that you use KDE. Why am I asking? QLog does not store passwords itself, but uses an external Secure Store library (QTKeyChain) and Secure store of the platform. In the case of KDE, it is KWallet, in the case of Gnome it is the Password&Keys application. Therefore I am asking.

John-WE1DER, 2023-05-22:
I'm using the Archcraft distribution of arch and have openboxwm for the window manager. Not sure if I have any kde services running. I know I have qt and gtk for various other uses. However, I now know that I need to configure a secure storage library. I will do some digging into the arch wiki and repositories to see what I can come up with this evening and report back.
Thank you for your time and energy you are putting forth to help answer this!

John-WE1DER, 2023-05-23:
Got this resolved! I installed the gnome keyring package and now the password storage is working perfectly! I will post a comment to the AUR page for QLog and suggest adding a package dependency for GNOME-Keyring to the PKGBUILD file. Thank you very much for your help and pointing me in the right direction!
73's DE WE1DER

foldynl, 2023-05-24:
I have to verify how the password is stored when gnome-keyriing or kwallet is not present. QLog uses qtkeychain lib to store passwords to the secure stores. This library should contain a mechanism that supports to store a password even if kwallet is not installed.

Therefore I am reopening the issue.

I'm glad you solved it

ea5wa, 2023-06-11:
Testing LoTW Upload/Download working perfectly. I'll try to test it with ClubLog

ea5wa, 2023-06-11:
Clublog also tested. Working perfectly

foldynl, 2023-06-12:
Thanks for reporting. I leave this issue in "need test" because i have to test it under OS where no password storage is configured. Under this situation, QLog has to be able to store password in a encrypted file. It is untested situation therefore I leave it open now.

foldynl, 2023-06-12:
I did some research and found this:

- qtkeychain supports Secret Storages:  LibSecretKeyring, GnomeKeyring, Kwallet4, Kwallet5 under Linux
- Everything works smoothly if system contains GnomeKeyring, Kwallet4 or Kwallet5
- I am not able to setup LibSecretKeyring (via ??secret-tool??)

QLog needs to be extended to report errors from Secret Storage. The extended error handling will be a part of a next release.

closed 2023-06-12 by foldynl