foldynl/QLog #204: Wrong QSO time saved

mppks, 2023-05-24:
Hi, when I'm working in ft8 (JTDX) and saving the QSO data, the wrong time is saved in the log - a difference of 3 hours, as if it is being converted to UTC again.
Link to the video with the problem.
https://drive.google.com/open?id=11_Ob0XEARxAYFYbIS2G9q8Aq-F19YeCn&usp=drive_fs
[image]

foldynl, 2023-05-25:
I did some research with these findings

1) It is an isolate issue related only to JTDX. WSJT-X is not impacted
2) If I look at the network messages then I can see that JTDX sends a different message format. Especially in cased of Date - JTDX sends time without Timezone information what is interpreted as Local timezone. Qlog then recalculates the local timezone to UTC timezone what causes what you see. WSJTX sends date with the timezone information.

It seems to be an issue of JTDX that the message is not consistent with the WSJTX message format. It also seems that JTDX is not longer developed (last update on 2022-03-01)  therefore  it will be hard to open an issue report with JTDX team.

That's why I will add "specific fixes" to QLog for JTDX. The fixes will be a part of next release.

foldynl, 2023-05-25:
Just attaching the messages received from WSJTX and JTDX (version 2.2.159)

```14:45:00.300 [DEBUG   ] [0x7f6cf1455480]                          [qlog.core.wsjtx.runtime] => "WsjtxLog: (ID: WSJT-X; DateTimeOff: Thu May 25 12:44:58 2023 GMT; DXCall: OK1AA; DXGrid: ; TXFreq: 14079500; Mode: JT9; RrpSent: ; RrpRcvd: ; TxPower: ; Comments: ; Name: ; DateTimeOn: Thu May 25 12:44:58 2023 GMT; OpCall: ; MyCall: OK1MLG; MyGrid: JO70AA; ExchSent: ; ExchRcvd: ; ADIFPropMode: ; )" [void Wsjtx::readPendingDatagrams():core/Wsjtx.cpp:345] ```

```14:52:48.130 [DEBUG   ] [0x7f6cf1455480]                          [qlog.core.wsjtx.runtime] => "WsjtxLog: (ID: JTDX; DateTimeOff: Thu May 25 12:53:00 2023; DXCall: OK1AA; DXGrid: AA11AA; TXFreq: 14075500; Mode: FT8; RrpSent: -15; RrpRcvd: -15; TxPower: ; Comments: ; Name: ; DateTimeOn: Thu May 25 12:52:26 2023; OpCall: OK1MLG; MyCall: OK1MLG; MyGrid: JO80AA; ExchSent: ; ExchRcvd: ; ADIFPropMode: ; )" [void Wsjtx::readPendingDatagrams():core/Wsjtx.cpp:345] ```

mppks, 2023-05-25:
> I did some research with these findings
>
>     1. It is an isolate issue related only to JTDX. WSJT-X is not impacted
>
>     2. If I look at the network messages then I can see that JTDX sends a [...]

OK thanks

commit 076f95ae (2023-05-28): Fixed #204 - Wrong QSO datetime in case of JTDX
closed 2023-06-09 by foldynl