foldynl/QLog #217: QSOs imported from DX Cluster widget doesn't import the Mode

ea5wa, 2023-06-13:
Double-click on DX Cluster widget, takes the callsign and frequency to the NewContactWidget but Mode/Missing is missing. This can lead to errors in the log if user don't realize.

foldynl, 2023-06-14:
Yes, it is desired functionality. DX Cluster does not provide any reliable information about the Mode. The only reliable information is the frequency. QLog has built-in frequency conversion to PHONE/DIGI/FT8/CW, but the Mode cannot be clearly determined based on this. Therefore, QLog does not transfer Mode information to NewContact.

I know only one reliable way to determine - connect the RIG via CAT.

Please, if you know any algorithm how to determine Mode based on frequency, region (QTH) and other parameters, it will be welcome.

ea5wa, 2023-06-14:
What do you mean about "reliable"? Although is not reliable is better than nothing, isn't it?

commit ab671b66 (2023-12-10): Fixed #217 - DXC: Rig Mode is switched based on the spot