foldynl/QLog #219: Multi language Support

dl2ki, 2023-06-17:
The wiki lacks examples for use on Windows and Mac OS.

foldynl, 2023-06-19:
please, could you specify what is missing in the description? The map language is controlled the same way as the GUI language.

dl2ki, 2023-06-19:
Sorry. I meant the wiki page "Multi language Support".

Here is an example how to start QLog with a different language translation.

> The language can also be set manually via OS's locales. An example below shows how the language can be changed manually to Czech via OS environment setting (Linux):
> 
> LANGUAGE=cs.CZ.UTF-8 ./qlog

My thought was that a Windows or Mac OS user might also need an example. Since I don't use Windows or Mac OS I can't say anything more about this.

foldynl, 2023-07-05:
Wiki has been updated.Closing issue.

closed 2023-07-05 by foldynl